hjam40/Camera.MAUI #2: Black screen with an Android device (s21 ultra)

PamperedBytes, 2023-03-24:
Nice project (works fine under windows by following the mini tuto on the github page), but I have a black screen when I launch the app with an Android device (Samsung galaxy s21 Ultra exynos version). Strange thing, when my app switches to thumbnail (by pressing the home button on my phone), for one frame I see the camera stream, before everything goes black again.

hjam40, 2023-03-26:
Hi, thanks for your feedback!
I have tested this project in 4 android devies and it works fine, but I have not a Samsung device for testing...
Maybe you could clone this git and debug if any exception is throwed in your device.
Any camera works ? (back/front)

jfversluis, 2023-04-03:
I'm seeing something similar here. When I use the sample app in this repo I have to start the camera, then stop it and then start it again for it to actually start working. 

Also a Samsung but a Galaxy S10e

luke-barnett, 2023-04-03:
> I'm seeing something similar here. When I use the sample app in this repo I have to start the camera, then stop it and then start it again for it to actually start working.
> 
> Also a Samsung but a Galaxy S10e
> 
> Calling Stop before Start also works somewhow:
> 
> ```cs
> //await cameraView.StartCameraAsync();
> await cameraView.StopCameraAsync();
> await cameraView.StartCameraAsync();
> ```

I have the same experience with a OnePlus 6T

taublast, 2023-04-05:
> I'm seeing something similar here. When I use the sample app in this repo I have to start the camera, then stop it and then start it again for it to actually start working.
>
> Also a Samsung but a [...]

The actual fix is:

private async void CameraView_CamerasLoaded(object sender, EventArgs e)
    {
        await Task.Delay(500); //do not start camera until page navigation completes
        if (cameraView.Cameras.Count > 0)
        {
            cameraView.Camera = cameraView.Cameras.First();
            MainThread.BeginInvokeOnMainThread(async () =>
            {
                if (await cameraView.StartCameraAsync() == CameraResult.Success)
                {
                    controlButton.Text = "Stop";
                    playing = true;
                }
            });
        }
    }

Slm0nB, 2023-04-09:
> The actual fix is:
>
> ```
> private async void CameraView_CamerasLoaded(object sender, EventArgs e)
>     {
>         await Task.Delay(500); //do not start camera until page navigation completes
>         if [...]

I am using a Google Pixel 7 Pro and have seen the same issue "No Preview". I can confirm that adding the delay to CameraView_CamerasLoaded solves the issue.

hjam40, 2023-04-09:
Hi,

I have published a new realease 1.3. It changes the android handler and this problems must be resolved. Also It adds recording video feature.

closed 2023-04-12 by hjam40

jfversluis, 2023-04-12:
Very cool @hjam40 thank you! I will check it out!

MCvel, 2023-08-10:
Hi guys, 

first thing first, thanks to @hjam40 for the tool its awesome,

however, I'm facing this issue of the black screen, but only the second time I open the CameraPage, where I'm using the camera View, 

So, the first time it opens perfectly, I take a photo and then pass the byte array to another Page where user can see the photo taken, everything good until this step, but, if I go back (to camera page) I got the black screen, no matter if I go back to Main page and open CameraPage again, black screen remains and there's no way I can make it work again.

this is the navigation flow:
MainPage
    CameraPage (take picture and pass byte array image)
         PhotoDetailPage

I already tried the delay, it didn't work for me; and is pretty weird because above the black screen I have the Take Photo button, and if I clic it it takes the photo correctly, is just like the cameraview his hidden somehow, but I tried, seting a fix height and width, isVisible and all you can imagine... 

any clues?
 
also, do you guys have anything in the OnAppearing and/or OnDissapearing to handle this back and forth?

thank You!