hjam40/Camera.MAUI #3: "Access Error"

JasonYangNWI, 2023-03-28:
Hi, Thanks for the project. It's a good choice before .NET Maui community toolkit add the cameraview.
I tried it on my Samsung tablet, and got "access error" after calling "StartCameraAsync".
What is the reason of "access error"?

hjam40, 2023-03-28:
Hi! Thank you for your feedback. 
This error could be for different reasons... Maybe you has not set the Camera property before call StartCameraAsync? You can try to set MirroredImage property to True before call StartCameraAsync, in android thats changes the PreviewView implementation mode to compatible and maybe it works better in Samsung devices...
Another reason could be an exception when the plugging tries to open the camera, you could clone this git repository and run in debug mode the test program included in your device

JasonYangNWI, 2023-03-29:
Hi Hector,
Thanks for the quick reply.
I found the reason. That's because I use ZXing.net.maui to process the barcode on the page before I use Camera.naui.
Apparently, ZXing.net.maui doesn't stop the barcodereaderview automatically, and I didn't find any function to stop it.
Do you have the experience on ZXing.net.maui? Do you know how to stop the barcodereaderview?

After I successfully open the camera, I am going to take a snapshot and save it to a variable.
And then I am going to show it on the image control of next page.
To do that, I call "GetSnapShot". As it returns "ImageSource", I must convert it to Base64 string.
Here is the code.
        var imageSource = cameraView.GetSnapShot(ImageFormat.PNG);
        var imageSourceStream = await ((StreamImageSource)imageSource).Stream.Invoke(new CancellationToken());
        using var memoryStream = new MemoryStream();
        imageSourceStream.CopyTo(memoryStream);
        weighRecord.Picture = Convert.ToBase64String(memoryStream.ToArray());
        
On the next page, I convert Base64 string back to image source.
        var imageBytes = Convert.FromBase64String(weighRecord.Picture);
        using MemoryStream memoryStream = new(imageBytes);
        contaminationPicture.Source = ImageSource.FromStream(() => memoryStream);
        
Unfortunately, it doesn't work. Do you know how to show the snapshot in the image control? 
Thank you very much.

hjam40, 2023-03-29:
Hi Jason
I know ZXing.net.maui, but this plugging has ZXing.Net integrated too and maybe you don't need the first. You can detect barcodes with the CameraView using the event "BarcodeDetected" and setting the property "BarCodeDetectionEnabled" to True (if you set it to False, the barcode detection stops) and "BarCodeOptions" property.
For image manipulation I recomend you to use the method "SaveSnapShot", It saves the image to a file and then you can read the bytes from the file and convert it to Base64 (wihtout ImageSource).
If you wanna use your code, I think that you have to set de MemoryStream Position to 0 before use it in the ImageSource:
var imageBytes = Convert.FromBase64String(weighRecord.Picture);
using MemoryStream memoryStream = new(imageBytes);
**memoryStream.Position = 0;**
contaminationPicture.Source = ImageSource.FromStream(() => memoryStream);

JasonYangNWI, 2023-03-30:
Hi Hector,

I fixed the image issue. It’s caused by “using” statement.

I am going to replace ZXing.net.maui with your Camera.maui plugin.
Some suggestions to Camera.maui plugin. Is it possible to make it compatible to MVVM pattern? Instead of calling function directly from the control, could you make some bindable property or event, which can be converted to command easily?

Thanks for the help once again. 
You can close the issue now.

Jason

closed 2023-03-30 by JasonYangNWI

hjam40, 2023-03-30:
Hi Jason,

Your suggestion is welcomed. I added bindable properties for the snapshotting and barcode detection.