hjam40/Camera.MAUI #4: Barcode Reader keeps working until BarCodeDetectionEnabled is set to false

JasonYangNWI, 2023-03-31:
Hi Hector,

I tried barcode reader function today. It's very good.
Just one issue.
After I turn on BarCodeDetectionEnabled, the engine keeps working. 
As I turn off BarCodeDetectionEnabled inside the event handler, usually my code will receive multiple barcode detected event.
Is there any way to detect only once?

Thank you very much.

Jason

hjam40, 2023-03-31:
Hi Jason,

This happens because the event handler runs in diferent tasks for performace reasons and I supose that new tasks are init before you change the BarCodeDetecionEnabled property. You could increase the BarCodeDetectionFrameRate property and the engine will have time to read the property before raise a new event.

JasonYangNWI, 2023-04-02:
Hi Hector,

I changed the rate to 20, and then 30(I believe the default rate is 10). 
There is no much difference. The event was always triggered twice.

And about MVVM pattern, thanks for the effort.
But I don't think I can get rid of code behind file 100%. 
As long as I must call "StartCameraAsync", "StopCameraAsync" and "GetSnapShot", I can't move them into view model.

Jason

hjam40, 2023-04-04:
Hi Jason,

It is possible to do everything by MVVM. In the next push, you will see an example code with a page using ViewModel and the control.

taublast, 2023-04-06:
> Hi Hector,
> 
> I tried barcode reader function today. It's very good. Just one issue. After I turn on BarCodeDetectionEnabled, the engine keeps working. As I turn off BarCodeDetectionEnabled inside the event handler, usually my code receives multiple barcode detected events. Is there any way to detect only once?
> 
> Thank you very much.
> 
> Jason

you could implement a lock for your event handler like a Semaphore and such to process only one event at a time, you can check the time between events to ignore those too close to eachother, but would first check if you're not subscribing several times.
when in doubt it is always a good practice to unsubscribe before subscribing
```
//Safe subscribing
control.EventOfInterest -= MyHandler;
control.EventOfInterest += MyHandler;

```this could be a quick fix.

JasonYangNWI, 2023-04-06:
Thanks, @taublast. Very helpful.
I did unsubscribe and subscribe event. That's not an issue for me.
But using a semaphore to lock the event handler is a good idea. I chose mutex and solved the problem. Thank you very much.

closed 2023-04-06 by JasonYangNWI

hjam40, 2023-04-09:
Hi Jason,

The new realease has an example of full MVVM page/model working with the control.