hjam40/Camera.MAUI #7: Issues working with Blazor MAUI Projects.

nathanpriceAMX, 2023-04-04:
Ive been trying to use you're Camera.MAUI view in my Blazor MAUI project, however I can't seem to get it to work and all the examples here use XAML. 

Ive followed the examples best ive can and making adjustments for Blazor. Which includes adding `.UseMauiCameraView` to "Mauiprogram.cs", giving the Android device Camera permissions and adding `xmlns:cv="clr-namespace:Camera.MAUI;assembly=Camera.MAUI"` to `MainPage.xaml`

However, the bit that i get an error on is: `cameraView.Camera = cameraView.Cameras.First();` which is part of the function ran when the button is pressed to display the camera. 

Where the issue is that `cameraView.Cameras` is never set, which is caused by the `CameraViewHandler` never being run and since `CameraViewHandler` is Protected i'm not able to call it myself. 

Would you know how to address this?

taublast, 2023-04-04:
My guess is you have to use camera view over your blazor app view, so that it picks up current platform handler.

hjam40, 2023-04-04:
Hi Nathan

This plugging is based on MAUI handlers for platforms, and It does not work in blazor webview (I don't have much knowledge about blazor) but as @taublast  has said, you coluld use it in a independent xaml page, out of the Webview.

nathanpriceAMX, 2023-04-05:
> Hi Nathan
> 
> This plugging is based on MAUI handlers for platforms, and It does not work in blazor webview (I don't have much knowledge about blazor) but as @taublast has said, you coluld use it in a independent xaml page, out of the Webview.

Okay, thanks for your help anyway. I'll give that a look into.

AOgai, 2023-04-05:
Guten Tag,

Gerald made a very good video about this topic:
[3 Ways Combine .NET MAUI and .NET MAUI Blazor Hybrid](https://www.youtube.com/watch?v=2dllz4NZC0I)

nathanpriceAMX, 2023-04-05:
> Guten Tag,
> 
> Gerald made a very good video about this topic: [3 Ways Combine .NET MAUI and .NET MAUI Blazor Hybrid](https://www.youtube.com/watch?v=2dllz4NZC0I)

Danke Sehr, 

That video was super helpful and ive now got it working in my application and i'm able to swap between it and my Blazor app.

closed 2023-04-05 by nathanpriceAMX