hjam40/Camera.MAUI #8: Cameras not being detected on published app, works just fine when build set to Debug.

MosheMalatji, 2023-04-05:
I am using the camera view to take pictures, it works well when I set the build to Debug however as soon as I publish the app and change build to Release the app crashes. It's almost as if it no longer detects the cameras.  I have tried it with this sample (https://github.com/jfversluis/MauiCameraMauiSample) as well with the same issues, is there something I am doing wrong ? 

See screenshots attached.

[image]

[image]

taublast, 2023-04-06:
It looks like you're missing camera permissions inside your manifest. In case you are compiling versus api 33 (android:targetSdkVersion="33"):

<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"  android:maxSdkVersion="28"/>
<uses-permission android:name="android.permission.CAMERA" />

Btw you could connect you device and launch release build while having been typed in terminal "adb logcat" to see in logs why you are crashing.

Now to be able to save your photos You'd need to manually request Storage access permissions for android below 10 only. Those above do not need them to save and read their own photos (compiling vs api >= 33 only), moreover if you request this permission for android 13 you will not get a prompt and the permission will always be refused, so..

MosheMalatji, 2023-04-06:
Thank you for your suggestion I will check the logcat however I can confirm that I am requesting camera permissions, the dialog does appear at the beginning, I've tested on 2 devices a Poco M3 Pro(Android 12) and a Xiaomi Redmi 8A(Android 11)  with the same result.
 
see android manifest below
```
<manifest xmlns:android="http://schemas.android.com/apk/res/android">
	<application android:allowBackup="true" android:icon="@mipmap/icon" android:roundIcon="@mipmap/icon_round" android:supportsRtl="true">
		<provider android:name="androidx.core.content.FileProvider"
          android:authorities="${applicationId}.fileprovider"
          android:exported="false"
          android:grantUriPermissions="true">

			<meta-data android:name="android.support.FILE_PROVIDER_PATHS"
						   android:resource="@xml/file_paths"></meta-data>
		</provider>
	</application>
	<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
	<uses-permission android:name="android.permission.INTERNET" />
	<uses-permission android:name="android.permission.CAMERA" />
	<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
	<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
	<uses-feature android:name="android.hardware.location" android:required="false" />
	<uses-feature android:name="android.hardware.location.gps" android:required="false" />
	<uses-feature android:name="android.hardware.location.network" android:required="false" />	<queries>
		<intent>
			<action android:name="android.media.action.IMAGE_CAPTURE" />
		</intent>
	</queries>
</manifest>

```

PeterVanVogelpoel, 2023-04-06:
I have the same issue. Camera not running in Release.

taublast, 2023-04-06:
What is the android version of devices that are failing? 
What is android sdk you are compile with?

Again, if you are compiling vs api33 (android 13) remove 
`<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE">`
from manifest, use only
```
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"  android:maxSdkVersion="28"/>
<uses-permission android:name="android.permission.CAMERA" />
```
if you are compiling for lower api you must have both read and write permissions declarations inside manifest.

Now about the code inside app. You have to check current android version using DeviceInfo. If it's android >=13 do nothing.
If it's lower android:
A - If you compiled vs api33:
ask user for permissions Permissions.StorageWrite. if this device android < 10
B - If you compiled vs lower api
ask user for permissions Permissions.StorageWrite. if this device is running < 13

Again you have to *manually* ask user for storage write permissions, having declaration in the manifest is not enough. cameraview will not do that for you, and such code is missing in the example app at the moment.

taublast, 2023-04-06:
1 on which version of android does crash, is it real device or emulator?
2 have you implemented any code for Permissions.StorageWrite?

PeterVanVogelpoel, 2023-04-07:
It happens on my device Samsung  Galaxy Note20 Ultra (Android 13).
Read or Write_External_Storage makes no difference. (Not need it anyhow)
I don't know how it works with adb log.

And again. Your Camera.MAUI.Test has the same problem. In Debug all fine. In Release camera not woring.

PeterVanVogelpoel, 2023-04-07:
Also (https://github.com/jfversluis/MauiCameraMauiSample)  has the same problem.

taublast, 2023-04-07:
1 this this not my repository
2 does your error happen when you click on Start button or when you open the page with camera? If it happens when you click on button my quick guess you just have to wrap all calls to cameraView.StartCameraAsync(); with a MainThread.BeginInvokeOnMainThread, that could matter if on Release the optimized code launches "async void" handlers in back thread, then calls camera permissions and crashes because this must be done on ui thread only.

PeterVanVogelpoel, 2023-04-07:
In Camera.MAUI.Test (Release) nothing happens when click on Start Camera button. In Debug all fine.

taublast, 2023-04-07:
My guessing was wrong, for me cameraView.Cameras are just empty on Release. The problem looks heavy.
[image]

PeterVanVogelpoel, 2023-04-07:
I don't need to release tomorrow. So I can continue developing and debugging.
I hope you are able to find the issue. I very much appreciate your repository.

quanzhenying, 2023-04-08:
Experiencing the same problem，debug mode was ok，but RC version failed.
Log: 'cameraView.Cameras.Count' is 0

hjam40, 2023-04-08:
Hi all,
I’m changing the entire android handler for the new release and adapt it to Camera2 and not CameraX (as now). I hope that it resolves all android problems.

AmmarAbo, 2023-04-08:
Hi All 
when do you expect this problem will be solving solve?
Thanks

hjam40, 2023-04-09:
Hi,

I have published a new realease 1.3. It changes the android handler and this problems must be resolved. Also It adds recording video feature.

PeterVanVogelpoel, 2023-04-09:
Great! It seems to work perfectly. Only the Zoomfactor seems not to work. Is that so?

hjam40, 2023-04-09:
Hi Peter,

with the new handler, zoom only works on API 30 or greater (at least for the moment).

PeterVanVogelpoel, 2023-04-10:
I'm using API 33, but when I add ZoomFactor="2" to the cameraView is has no effect.

PeterVanVogelpoel, 2023-04-10:
And after testing a bit more I experience some unexpected behaviour. I'm switching between different pages with a cameraView and sometimes the page is crashing. Still looking for the issue.

PeterVanVogelpoel, 2023-04-10:
Hi Hjam40, I'm using API 33, but the ZoomFactor seems to have no effect.
And another thing is that it seems that the scanner does not respond when reading the same QR Code for the second time. Is that intended?

hjam40, 2023-04-10:
Hi Peter,

I'm trying the ZoomFactor in all platforms and it works fine for me... Could you verify the MaxZoomFactor for your selected Camera?

In reference to the scanner, yes, I changed the event behaviour to trigger only if the result is different from the previous but I'm thinking that maybe is not a good idea and I will create a property for set the behaviour as the user prefers.

PeterVanVogelpoel, 2023-04-10:
Hi hjam40, sorry to bother you about the ZoomFactor. Wrong coding on my end. Problem solved.
And yes, I prefer that the scanner can read the same code multiple times.
One more question: I have a square frame for my cameraView (300 x 300). But then the image is stretched in height. Is there a way to solve that?

PeterVanVogelpoel, 2023-04-10:
And sometimes I'm getting:
Android.Hardware.Camera2.CameraAccessException: 'CAMERA_ERROR (3): endConfigure:740: Camera 0: Error configuring streams: Broken pipe (-32)'
Is this my code or something else?

PeterVanVogelpoel, 2023-04-10:
[image]

PeterVanVogelpoel, 2023-04-10:
Is there a way to detect video resolution selected by Android?

PeterVanVogelpoel, 2023-04-10:
And I'm using StopCameraAsync. I don't find StopPreviewAsync.

hjam40, 2023-04-10:
Sorry, my mistake, it's correct StopCameraAsync.

No way (at least I didn't find it) to know the Camera video resolution using in a preview in Android....

TT702, 2023-04-10:
> 

Having the similar issue after updating to 1.3. 
The preview seems to be weirdly stretched.

[image]
[image]

PeterVanVogelpoel, 2023-04-11:
Hi, I got it all working fine now. Thanks for the hard work.  
Except for the stretching of the preview. This was not there in the previous version. I hope this can be resolved.

MosheMalatji, 2023-04-11:
Thanks for the fix Hector, it works now. Keep up the good work.

closed 2023-04-11 by MosheMalatji

hjam40, 2023-04-11:
Hi,

I have created a new realease 1.3.1 where I have changed the aspect ratio for Android Preview... I hope it works for you.

PeterVanVogelpoel, 2023-04-11:
I updated, but then I get:
System.InvalidCastException: 'Specified cast is not valid.'

PeterVanVogelpoel, 2023-04-11:
After update I'm getting:
System.InvalidCastException: 'Specified cast is not valid.'
at StartCameraAsync

hjam40, 2023-04-12:
Hi Peter,

Sorry, but I can't get that error... Could you give me more information? Maybe the line where the exception throws.. You could clone the git repository and run the app on your device.