hjam40/Camera.MAUI #10: No proview / no snapshot

FrKaram, 2023-04-06:
Hi

I installed folowing the documentation but I get no preview in the CameraView control.
GetSnapShot() return a null ImageSource.

Any idea ?

MosheMalatji, 2023-04-08:
any chance you can share your code FrKaram so we can see what might be the problem ?

FrKaram, 2023-04-08:
Hi Moshe,

sure !
It is the most basic app possible.
[MauiApp4.zip](https://github.com/hjam40/Camera.MAUI/files/11182908/MauiApp4.zip)

MosheMalatji, 2023-04-09:
Hi Fikram I found that adding `await cameraView.StopCameraAsync();` before you call the `await cameraView.StartCameraAsync();` method helps,  especially if you aren't manually starting the camera, there might be a better workaround but try that for now.

Your CamerasLoaded method would look like this.

  ```
  private void CameraView_CamerasLoaded(object sender, EventArgs e)
    {
       if (cameraView.Cameras.Count > 0)
        {
            cameraView.Camera = cameraView.Cameras.First();
            cameraView.FlashMode = FlashMode.Auto;
            
            MainThread.BeginInvokeOnMainThread(async () =>
            {
                await cameraView.StopCameraAsync();
                if (await cameraView.StartCameraAsync() == CameraResult.Success)
                {
                }
            });

        }
    }
```

FrKaram, 2023-04-09:
Indeed it does work.
Thanks

closed 2023-04-09 by hjam40