hjam40/Camera.MAUI #12: After calling "StartCameraAsync", I got System.InvalidCastException on v1.3.1

JasonYangNWI, 2023-04-12:
After calling "StartCameraAsync", I got System.InvalidCastException on v1.3.1.
I have to revert back to v1.3, which is good.

hjam40, 2023-04-12:
Hi Jason,

Sorry, but I can't get that error... Could you give me more information? Maybe the line where the exception throws.. You could clone the git repository and run the app on your device.

JasonYangNWI, 2023-04-12:
Hi Hector, I didn't change any code. 
The same code work with v1.1 and v1.3, but not v1.3.1.
Here is the line. 
_var result = await cameraView.StartCameraAsync();_

hjam40, 2023-04-13:
Hi,

I have published package version 1.3.3. It fix Android preview and zoom problems (now zoom is again compatible with all API Versions supported by the package).

closed 2023-04-13 by hjam40