hjam40/Camera.MAUI #14: AutoFocus?

rvonkriegenbergh, 2023-04-13:
The lack of auto focus keeps us from using this. Does the ZXing control have an AutoFocus method? If so, can you expose it in this control so I can call it periodically? If this is not possible, is there another way to force a focus adjustment?

hjam40, 2023-04-13:
Hi Roland,

AutoFocus is set by default for IOS and Android Handlers. I'm not sure if anyone wants disable it for use it as a property. Maybe if you want force a focus, change the zoom factor trigger a focus in the camera.

Android example:
previewBuilder.Set(CaptureRequest.ControlMode, Java.Lang.Integer.ValueOf((int)ControlMode.Auto));

hjam40, 2023-04-14:
Hi Roland,

I have tested the application on IOS/Android and the camera focuses itself every few seconds if you move an object closer or further away. Does it work the same way for you?
I have been looking at the repository that you indicate and the Android AutoFocus function is empty, so it does nothing...

rvonkriegenbergh, 2023-04-14:
Thanks for looking into this.

Our current app is web based and uses Angular and a ZXing plugin to scan barcodes. We scan high volumes of barcodes in a greenhouse environment. Users complain about how slow the auto focus is. It makes it impossible to use for the high scan volume we have. So we are trying to find another solution. 

Your MAUI plugin works as you describe it. It does auto focus for me too, eventually, every few seconds when changing the distance between the barcode and the phone. But it is very cumbersome and too slow. The auto focus of the regular built-in camera is much faster. Not sure why ZXing can't be improved. Being able to tap into the view to focus on that distance (like regular camera apps) would also be helpful or at least expose an auto focus function that actually works.

Thanks for your great work making this work in MAUI but our search for a barcode scanner camera plugin, regardless of dev environment (web, MAUI, Flutter ...) continues.

hjam40, 2023-04-15:
Hi Roland,

Right now I understand the problem. In the new release (1.3.5) you can find ForceAutoFocus method, it forces the auto focus trigger in the camera. I have not could test it in IOS (at least for the next week). I hope it works fine for you.

rvonkriegenbergh, 2023-04-19:
Thank you for adding this! I added a button to the view under the camera view. Pushing it calls the ForceAutoFocus  method. It looks like it doesn't do anything. The focus still takes a very long time. Nothing really happens when I call that new method.

hjam40, 2023-04-25:
Hi Roland,

Could you record a video with the behaviour? I my devices if I have a unfocus preview image and call the function, the camera focus the image.

Light407, 2024-11-03:
Is there any news regarding this? We need to put the phone quite far away to get focus but other libs focus very clear on close distance.