hjam40/Camera.MAUI #15: Flip cameras not working

carolzbnbr, 2023-04-13:
Hi

I'm trying to flip (change) cameras on iOS device but it does not work.

The front camera works fine, but when I click the button to switch to back camera, it stop the camera stream.
I also tried start with the back camera, it works, but when I hit the button to switch to front camera it also stops the stream.

Here is the code:

```cs

    private void CameraView_CamerasLoaded(object sender, EventArgs e)
    {
        StartCamera(CameraPosition.Front);

    }

    private async void StartCamera(CameraPosition position)
    {
        await MainThread.InvokeOnMainThreadAsync(async () =>
        {
            await cameraView.StopCameraAsync();
            cameraView.Camera = null;

            cameraView.Camera = cameraView.Cameras.Where(f => f.Position == position).First();
            cameraView.MirroredImage = true;

            await cameraView.StartCameraAsync();
            canvas.InvalidateSurface();
        });
    }

    void CameraView_FlipCamera(System.Object sender, System.EventArgs e)
    {
        CameraPosition position = cameraView.Camera.Position == CameraPosition.Front ? CameraPosition.Back : CameraPosition.Front;
        StartCamera(position);
    }
```

and xaml:

```xml
    <cv:CameraView x:Name="cameraView"
                       Grid.RowSpan="4"
                        CamerasLoaded="CameraView_CamerasLoaded"
                                />

                   <Button
                   Grid.Row="0"
                    Grid.Column="2"
                    Text="{Binding Source={x:Static helpers:MaterialDesignIconFonts.CameraFlip}}"
                    Style="{StaticResource TreinifyIconLabelStyle}"
                    Clicked="CameraView_FlipCamera"
                    />
```

hjam40, 2023-04-14:
Hi Carol,

I have pushed a new version on master brand that maybe fix the problem, but I can't test it in iPhone for the next week. If you want, you could test the repo in your iPhone with the app included and if it works I will release a new version in nuget.

hjam40, 2023-04-18:
Hi Carol,

Have you been able to test if the problem is resolved with the latest version of the plugin?

carolzbnbr, 2023-04-19:
Hi Hector,

I didn't have the chance to grab your code locally and test it. 
I'm probably going to do that later this week and will get you back.

Anyway, thanks a lot for making the changes.

hjam40, 2023-04-19:
Hi Carol,

finally, you don't need to do that. The last package version has the change implemented.

hjam40, 2023-04-25:
Hi,

I test it and it works fine in the new version

closed 2023-04-25 by hjam40