hjam40/Camera.MAUI #18: Saved .Jpg files are limited resolution

pierre01, 2023-04-19:
The saved files using the stream returned from  CameraView.TakePhotoAsync();
Are limited to 788x1054 on all the cameras listed on my Samsung S22 Ultra. (front and back)

Can we get an image with the full resolution of the selected camera?

Using version 1.3.5

hjam40, 2023-04-19:
Hi Pierre,

Right now, Android photo resolution is limited to CameraView control size and device screen resolution, but I'm working on changing it. You could create a full screen control and the capture resolution should be higher.

hjam40, 2023-05-12:
Hi Pierre,

In the new version you could set a Resolution in StartCameraAsync. This resolution will be used by TakePhotoAsync.

FrKaram, 2023-05-12:
Gonna try that immediately, thanks

closed 2023-05-15 by hjam40

pierre01, 2023-05-23:
Thank you for fixing this so fast. Thank you for your magnificent work ... It makes the camera finally usable without a picker.

hjam40, 2023-05-23:
Hi Pierre, thank you for your support!