hjam40/Camera.MAUI #19: no preview

SalimiHabib, 2023-04-20:
work fine but with front camera , after switch to back camera have  no preview .
I use provided test project in GitHub  with no change , just start it .
test phone is  honor 8A  android 9 

also  this in error in output  [GLConsumer] [SurfaceTexture-0-20313-0] bindTextureImage: error binding external image: 0x502

hjam40, 2023-04-21:
Hi Salim,

It looks like a internal mono android error and I can't do nothing to solve it. Maybe the problem is the API version (so old).

SalimiHabib, 2023-04-23:
Android 9 is old but not so old , also this library work by front camera but back camera no preview and no error from library  ,  if there  is minimum version restriction for this library please specify ?
Thank you

SalimiHabib, 2023-04-29:
found this may be it can help 

https://stackoverflow.com/questions/27626002/camera-preview-using-surface-texture

Maryusz, 2023-05-10:
Same error found here with Android 9 on a ZTE phone, works correctly on other phones but on this one it causes:

[GLConsumer] [SurfaceTexture-0-19290-0] bindTextureImage: error binding external image: 0x502
[IMGSRV] :0: IsTextureConsistent: IMGEGLImage is not consistent