hjam40/Camera.MAUI #20: No Windows Preview nor Windows Barcode Detection

DavidPressman, 2023-04-21:
Using Gerald Versiuis' two recent Camera.View sample projects, as discussed in his matching YouTube videos, I am unable to see a CameraView preview in either when using a Windows 11 Webcam and running VS 2022 Enterprise Preview. (I had to make minor changes to his csprog files as neither contained a properly-formatted ApplicationIdGuid and an assembly neutral language, both required for Windows.) However, in the earlier project for image capture, an image can be captured from that webcam and properly displayed beneath where the preview should appear. In the latter project for barcode scanning, the same webcam does get started (still no preview), but no barcode detection occurs. Same project works as expected on an actual Android device.

SalimiHabib, 2023-04-23:
it seems issues is the same , check it #19

araghon007, 2023-04-24:
It might be related to #2 
I had to add the 500 ms delay in order to get a preview working in the latest version on Windows. Haven't yet tested on other platforms.

DavidPressman, 2023-04-24:
Yes, adding a delay solved the preview issue. I would suggest the StartCameraAsync() function be amended as StartCameraAsync(int delay = 0) with "if (delay > 0) await Task.Delay(delay)" as first line.
Not sure if Camera.Maui can actually detect a Barcode in Windows. I haven't been able to, but could be a focus / resolution issue. Would love to know what barcode its author used to test, as it is based on ZXing.Maui, which specifically says it cannot yet detect barcodes in Windows (see https://github.com/Redth/ZXing.Net.Maui).

hjam40, 2023-04-25:
Hi David,

I understand that the problem only happens the first time you start the preview when the control is loaded. It may be because the event indicates that the handler has been attached is fired before the native control is ready and therefore the transmission does not start. I'll look into adding a delay on the control to indicate it's ready.

Regarding the barcode detection, I have tested the QR detection in Windows and it works fine. It's not as fast as mobile and you have to play with distance more, but that depends on the quality of the camera you're using.

closed 2023-05-15 by hjam40