hjam40/Camera.MAUI #25: PopAsync crashes on Android with "System.ObjectDisposedException: 'Cannot access a closed Stream.'"

Raffro, 2023-05-05:
I use the cameraView within a ContentPage, which is pushed async on a shell page.

When navigating back to the shell page it says
'Cannot access a closed Stream.'

This is the case when a snapshot has been taken with
* GetSnapShot() or
* TakePhotoAsync()
At this point I am not sure, what exactly it is meant by Stream, because there is no stream open ....

hjam40, 2023-06-01:
Hi,

Does this problem continue with the new versions? I can't reproduce it.

Raffro, 2024-03-19:
> Hi,
> 
> Does this problem continue with the new versions? I can't reproduce it.

I dont have the error anymore becauae I wrapped the whole photasynccapture method in try catch