hjam40/Camera.MAUI #28: Second (And only second!) photo after first run always fails. [Warning: Unable to load image stream.]

GarthConnell, 2023-05-10:
We have exactly the same issue with an app we are working on, so I figured lets see how others have got around it!
I just pull the repo and run it, click smile, and second click fails with log below:

[0:] Microsoft.Maui.StreamImageSourceService: Warning: Unable to load image stream.

Android.Util.AndroidRuntimeException: Only the original thread that created a view hierarchy can touch its views.
   at Java.Interop.JniEnvironment.StaticMethods.CallStaticVoidMethod(JniObjectReference type, JniMethodInfo method, JniArgumentValue* args) in /Users/runner/work/1/s/xamarin-android/external/Java.Interop/src/Java.Interop/obj/Release/net7.0/JniEnvironment.g.cs:line 13250
   at Java.Interop.JniPeerMembers.JniStaticMethods.InvokeVoidMethod(String encodedMember, JniArgumentValue* parameters) in /Users/runner/work/1/s/xamarin-android/external/Java.Interop/src/Java.Interop/Java.Interop/JniPeerMembers.JniStaticMethods.cs:line 97
   at Microsoft.Maui.PlatformInterop.LoadImageFromStream(ImageView imageView, Stream inputStream, IImageLoaderCallback callback) in D:\a\_work\1\s\src\Core\src\obj\Release
et7.0-android\generated\src\Microsoft.Maui.PlatformInterop.cs:line 443
   at Microsoft.Maui.StreamImageSourceService.LoadDrawableAsync(IImageSource imageSource, ImageView imageView, CancellationToken cancellationToken) in D:\a\_work\1\s\src\Core\src\ImageSources\StreamImageSourceService\StreamImageSourceService.Android.cs:line 28
  --- End of managed Android.Util.AndroidRuntimeException stack trace ---
android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
	at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:8798)
	at android.view.ViewRootImpl.requestLayout(ViewRootImpl.java:1606)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at androidx.drawerlayout.widget.DrawerLayout.requestLayout(DrawerLayout.java:1353)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at androidx.recyclerview.widget.RecyclerView.requestLayout(RecyclerView.java:4586)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at androidx.core.widget.NestedScrollView.requestLayout(NestedScrollView.java:2075)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.widget.ImageView.setImageDrawable(ImageView.java:597)
	at androidx.appcompat.widget.AppCompatImageView.setImageDrawable(AppCompatImageView.java:112)
	at com.microsoft.maui.glide.MauiCustomViewTarget.onResourceCleared(MauiCustomViewTarget.java:30)
	at com.bumptech.glide.request.target.CustomViewTarget.onLoadCleared(CustomViewTarget.java:210)
	at com.bumptech.glide.request.SingleRequest.clear(SingleRequest.java:337)
	at com.bumptech.glide.request.ErrorRequestCoordinator.clear(ErrorRequestCoordinator.java:48)
	at com.bumptech.glide.manager.RequestTracker.clearAndRemove(RequestTracker.java:70)
	at com.bumptech.glide.RequestManager.untrack(RequestManager.java:660)
	at com.bumptech.glide.RequestManager.untrackOrDelegate(RequestManager.java:628)
	at com.bumptech.glide.RequestManager.clear(RequestManager.java:624)
	at com.bumptech.glide.RequestBuilder.into(RequestBuilder.java:811)
	at com.bumptech.glide.RequestBuilder.into(RequestBuilder.java:780)
	at com.bumptech.glide.RequestBuilder.into(RequestBuilder.java:771)
	at com.microsoft.maui.PlatformInterop.prepare(PlatformInterop.java:229)
	at com.microsoft.maui.PlatformInterop.loadInto(PlatformInterop.java:234)
	at com.microsoft.maui.PlatformInterop.loadImageFromStream(PlatformInterop.java:265)

  --- End of managed Android.Util.AndroidRuntimeException stack trace ---
android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
	at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:8798)
	at android.view.ViewRootImpl.requestLayout(ViewRootImpl.java:1606)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at androidx.drawerlayout.widget.DrawerLayout.requestLayout(DrawerLayout.java:1353)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at androidx.recyclerview.widget.RecyclerView.requestLayout(RecyclerView.java:4586)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at androidx.core.widget.NestedScrollView.requestLayout(NestedScrollView.java:2075)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.view.View.requestLayout(View.java:25390)
	at android.widget.ImageView.setImageDrawable(ImageView.java:597)
	at androidx.appcompat.widget.AppCompatImageView.setImageDrawable(AppCompatImageView.java:112)
	at com.microsoft.maui.glide.MauiCustomViewTarget.onResourceCleared(MauiCustomViewTarget.java:30)
	at com.bumptech.glide.request.target.CustomViewTarget.onLoadCleared(CustomViewTarget.java:210)
	at com.bumptech.glide.request.SingleRequest.clear(SingleRequest.java:337)
	at com.bumptech.glide.request.ErrorRequestCoordinator.clear(ErrorRequestCoordinator.java:48)
	at com.bumptech.glide.manager.RequestTracker.clearAndRemove(RequestTracker.java:70)
	at com.bumptech.glide.RequestManager.untrack(RequestManager.java:660)
	at com.bumptech.glide.RequestManager.untrackOrDelegate(RequestManager.java:628)
	at com.bumptech.glide.RequestManager.clear(RequestManager.java:624)
	at com.bumptech.glide.RequestBuilder.into(RequestBuilder.java:811)
	at com.bumptech.glide.RequestBuilder.into(RequestBuilder.java:780)
	at com.bumptech.glide.RequestBuilder.into(RequestBuilder.java:771)
	at com.microsoft.maui.PlatformInterop.prepare(PlatformInterop.java:229)
	at com.microsoft.maui.PlatformInterop.loadInto(PlatformInterop.java:234)
	at com.microsoft.maui.PlatformInterop.loadImageFromStream(PlatformInterop.java:265)

hjam40, 2023-05-12:
Hi Garth,

I have no idea what is the problem with that... As you say, only the second time fails and I think that is an internal android problem (or mono droid). In the new plugging version, TakePhotoAsync works fine all times (because it use another android methods).

GarthConnell, 2023-05-12:
Yes, it is internal, and it shows a stream closing prematurely or disposing. Ill spend some time and work this one out!

pierre01, 2023-05-17:
You might want to declare the stream variable outside the function where you are using it. ( as a class member) because of all the asynchronous work happening, the locally declared variables might be disposed before the rest of the function code completes.

Hugol13, 2023-08-17:
@GarthConnell try put in your code a delay for 13000, before to star de camera, I had similar problem, but with this I fix it.

juliusdeblaaij, 2023-09-09:
This issue persists unfortunately, also when using AutoSnapShot functionality from XAML. Any workaround as currently I cannot use this lib on Android.