hjam40/Camera.MAUI #29: FlashMode is not working in android while adding FlashMode to Enabled.  cameraView.FlashMode = FlashMode.Enabled;

jyothiv7, 2023-05-12:
Hi,
When I was trying to assign the cameraView.FlashMode = FlashMode.Enabled; 
In iOS its is working fine and capturing photos with flash. But where as in Android, the flashmode is not working.
I tried to enable FlashMode as below.  
cameraView.FlashMode = FlashMode.Enabled;

hjam40, 2023-05-12:
Hi,
In the current version, android is not really taken a photo from camera, It takes a snapshot from preview. In the new version (comming soon) the behavior is the correct and the flash should works correctly.

closed 2023-05-15 by hjam40

jyothiv7, 2023-05-16:
Thanks for response. 
I have observed just the FlashMode is not working in Android. 

Thank you.

> Hi, In the current version, android is not really taken a photo from camera, It takes a snapshot from preview. In the new version (comming soon) the behavior is the correct and the flash should works correctly.

bzburro, 2024-09-18:
had this been fixed?