hjam40/Camera.MAUI #31: Problem with new build

mancda, 2023-05-12:
After upgrading to 1.4 I am now getting the following with testing the cameraview on my android device.  I have removed all my code so that the only think is the reference to the cameraview in the xaml .  

Here is the error 

{System.NullReferenceException: Object reference not set to an instance of an object.
   at Camera.MAUI.Platforms.Android.MauiCameraView.InitDevices()
   at Camera.MAUI.Platforms.Android.MauiCameraView..ctor(Context context, CameraView cameraView)
   at Camera.MAUI.CameraViewHandler.CreatePlatformView()
   at Microsoft.Maui.Handlers.ViewHandler`2[[Camera.MAUI.CameraView, Camera.MAUI, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null],[Camera.MAUI.Platforms.Android.MauiCameraView, Camera.MAUI, Version=1.4.0.0, Culture=neutral, PublicKeyToken=null]].OnCreatePlatformView() in D:\a\_work\1\s\src\Core\src\Handlers\View\ViewHandlerOfT.cs:line 72

mancda, 2023-05-12:
Tested version 1.4 on an iOS phone and it works fine.

SimonBrettschneider, 2023-05-15:
I have tried out version 1.4 on Android (google pixel 7 Pro) and have had similar issues. The app crashes. The older version works fine.

hjam40, 2023-05-17:
Hi,

sorry, but I can't reproduce the error. Could you download this repository and test the included app in debug mode?

mancda, 2023-05-18:
I have download the repository test app and found the line of code in MainCameraView.cs InitDevices that is causing the null exception.

                foreach(var s in map.GetOutputSizes(Class.FromType(typeof(ImageReader))))
                    cameraInfo.AvailableResolutions.Add(new(s.Width, s.Height));

Works fine on the pixel 5 emulator but errors out on my Samsung S20 device.

hjam40, 2023-05-19:
Hi,

Really I don't understand why it crash... I call the same function in other code part (in previews versions too) and It works... by the way I added an exception control in the last version (1.4.2) and It should works right now.

mancda, 2023-05-19:
That fixed it.  Thanks

closed 2023-05-19 by mancda