hjam40/Camera.MAUI #33: Crash on Android 8 (AudioManager  getMicrophones())

Arodos, 2023-05-13:
On a device with Android 8 (API Level 26) the app crashes with the method "CamerasLoaded or MicrophonesLoaded" with the error message:
"**Java.Lang.NoSuchMethodError:** 'no non-static method "Landroid/media/AudioManager;.getMicrophones()Ljava/util/List;"'"
The CamerView works on an emulator with Android 13.

hjam40, 2023-05-15:
Hi,

When do you get that error? In the last version of the plugging Android 8 does not read microphones...

Arodos, 2023-05-15:
After the update, only the AccessError came up

The CameraView is defined by: `<cv:CameraView x:Name="cameraView" CamerasLoaded="camerView_CamerasLoaded"/>`

When the page loads, the following code is executed:
```
private void camerView_CamerasLoaded(object sender, EventArgs e)
     {
             cameraView.Camera = cameraView.Cameras.First();
             MainThread.BeginInvokeOnMainThread(async () =>
             {
                 var result = await cameraView.StartCameraAsync();
                 lblCameraResult.Text = "Camera: " + result.ToString();
             });
     }
```

In the emulator (API 33) you can see the camera and the label says "Success". The device (API 26) only says "AccessError". If you disconnect the device from the computer or create a release version, the app stops or crashes.

This is the console output:

> [ViewRootImpl@4390c06[MainActivity]] ViewPostIme pointer 0
> [ViewRootImpl@4390c06[MainActivity]] ViewPostIme pointer 1
> [zygote64] Do full code cache collection, code=112KB, data=97KB
> [zygote64] After code cache collection, code=90KB, data=61KB
> [monodroid-assembly] open_from_bundles: failed to load assembly Microsoft.Extensions.Primitives.dll
> Loaded assembly: /data/data/com.companyname.mycloset/files/.__override__/Microsoft.Extensions.Primitives.dll [External]
> [monodroid-assembly] open_from_bundles: failed to load assembly System.Linq.Expressions.dll
> Loaded assembly: /data/data/com.companyname.mycloset/files/.__override__/System.Linq.Expressions.dll [External]
> [monodroid-assembly] open_from_bundles: failed to load assembly System.ComponentModel.Primitives.dll
> Loaded assembly: /data/data/com.companyname.mycloset/files/.__override__/System.ComponentModel.Primitives.dll [External]
> [CameraManagerGlobal] Connecting to camera service
> [VendorTagDescriptor] addVendorDescriptor: vendor tag id 3854507339 added
> [CameraManager] getCameraCharacteristics : cameraId = 0
> [CameraManager] getCameraCharacteristics : cameraId = 1
> [CameraManager] getCameraCharacteristics : cameraId = 0
> [CameraManager] getCameraCharacteristics : cameraId = 0
> [Surface] sf_framedrop debug : 0x4f4c, game : false, logging : 0
> [Choreographer] Skipped 91 frames! The application may be doing too much work on its main thread.
> [Surface] sf_framedrop debug : 0x4f4c, game : false, logging : 0

Is there a way to fix this problem?

hjam40, 2023-05-17:
Hi

The previous package versions were not compatible with Android 8. The new version 1.4.1 is compatible.

closed 2023-05-19 by hjam40