hjam40/Camera.MAUI #34: iPhone 14 Pro Focus Issue

raconn, 2023-05-13:
Using iOS 16.4, camera view loads along with barcode detection. However the camera view will not focus on a barcode in order to detect it. No exceptions thrown as in latest ZXing library.

raconn, 2023-05-13:
I did end up getting it to work on an iPhone 14 Pro, running iOS 16.4. It didn't really work out of the box, so I had to set the ZoomFactor to some arbitrary value and then call the ForceAutoFocus method. The bulk of this came from Gerald's youtube video on the library, except for the ZoomFactor and AutoFocus. Maybe this was in the documentation somewhere, but I didn't find it. Thanks!

```
<cv:CameraView ZIndex="1" x:Name="cameraView" WidthRequest="300" HeightRequest="300"
                           CamerasLoaded="CameraView_CamerasLoaded"
                           BarCodeDetectionEnabled="True"
                           BarcodeDetected="cameraView_BarcodeDetected" />
```

```
private void CameraView_CamerasLoaded(System.Object sender, System.EventArgs e)
    {
		if (cameraView.Cameras.Count > 0)
		{
			cameraView.Camera = cameraView.Cameras.First();
			MainThread.BeginInvokeOnMainThread(async () =>
			{
				await cameraView.StopCameraAsync();
				await cameraView.StartCameraAsync();
			});

            if (cameraView.MaxZoomFactor >= 3.5f)
            {
                cameraView.ZoomFactor = 3.5f;
            }
            cameraView.ForceAutoFocus();
        }
    }
    
void cameraView_BarcodeDetected(System.Object sender, Camera.MAUI.ZXingHelper.BarcodeEventArgs args)
    {
        MainThread.BeginInvokeOnMainThread(() =>
        {
            var barcodeFormat = args.Result[0].BarcodeFormat;
            var barcode = args.Result[0].Text;

            Console.WriteLine($"Barcode Format: {barcodeFormat} - Barcode: {barcode}");
        });
    }
```

hjam40, 2023-05-15:
Hi,

AutoFocus function is only needed it you want to force the autofocus. By default, the camera is opened with autofocus enabled. When I test it in iPhone (XS and 14) the camera autofocus when i change the image or the distance to the barcode

raconn, 2023-05-15:
Odd. I could not get it to autofocus until I provided a zoom factor and called the force autofocus method. Works fine with that sequence in my use case. 

Thanks for looking into it.

closed 2023-05-15 by raconn

hjam40, 2023-05-18:
Hi,

In the new version (1.4.1) I have changed the start camera behavior in iOS to set Continuous Auto focus by default, you could test it.

raconn, 2023-05-22:
Hello - I tested and cannot get the camera view to render at all now. I placed 2 frame layers over the camera view to indicate a center line for the camera view. The center line shows, but the camera view is black.

FrKaram, 2023-05-22:
One user notified me of a focus issue with an iphone 14 pro max (seems that the view is always out of focus)
Haven't tries version 1.4.1 yet to see if it solves the issue.
@raconn You closed this issue, does this mean version 1.4.1 works for you ?

FrKaram, 2023-05-25:
> I created a workaround in my use case where I set zoom factor and then force autofocus which I showed in the sample code. I did work for me which is why I closed it. However, the latest version of the library does not work at all on iPhone 14. Shows a black screen where camera preview should be.
> 
> I hope that makes sense. If you need more clarification, please let me know.

Makes sense, thanks.
Have you tried version 1.4.3 which seems to solve the issue of the component not working on IOS ?
Does it work for you on iphone 14 ? Does it solve the focus issue ? I do not have an iphone 14 to test

raconn, 2023-05-25:
Yep. Confirmed it works on iPhone 14 Pro using version 1.4.3 of the library. 

I appreciate your help!