hjam40/Camera.MAUI #35: android 8 noSuchMethodError

gokhangencer, 2023-05-14:
I have also comment out  the following lines but getMicrophones() method is not found error thrown. 
tested on emulator and test phone.

//if (cameraView.NumMicrophonesDetected > 0)
            //    cameraView.Microphone = cameraView.Microphones.First();

exception:
[image]

call stack:
[image]

thank you.

hjam40, 2023-05-15:
Hi,

In the last version of the plugging, microphones read in Android is limited to API>=30:
```c#
            if (OperatingSystem.IsAndroidVersionAtLeast(30))
            {
                cameraView.Microphones.Clear();
                foreach (var device in audioManager.Microphones)
                {
                    cameraView.Microphones.Add(new MicrophoneInfo { Name = "Microphone " + device.Type.ToString() + " " + device.Address, DeviceId = device.Id.ToString() });
                }
            }
``` 

But as I said in the other post, Android does not use microphones list for video recording. You don't need initialize it. It use the default microphone input in the device.

hjam40, 2023-05-17:
Hi

The previous package versions were not compatible with Android 8. The new version 1.4.1 is compatible.

closed 2023-05-19 by hjam40