hjam40/Camera.MAUI #40: Camera not working in IOS

jarvis10211, 2023-05-23:
camera not working in ios

UkeHa, 2023-05-23:
@jarvis10211, please provide more information when you open an issue. What device were you on, which iOS version, what did you try, what code did you use?

hjam40, 2023-05-23:
Hi, there is a problem with the last version on iOS devices, it will be fixed in the next one (at the end of this week). Right now you could use the previous version.

jarvis10211, 2023-05-23:
Previouse mean, version please?

hjam40, 2023-05-23:
1.4.0 should works fine on iOS

hjam40, 2023-05-24:
Hi

I have released a new version (1.4.3), iOS should work fine with this one.

SimonBrettschneider, 2023-05-24:
Hi, I just tried the new version (1.4.3), upgrading from 1.3.5, which works fine on most Android devices, and I have trouble getting it up and running on Android:

I get these error messages.

**System.MissingMethodException:** 'Method not found: System.Threading.Tasks.Task`1<Camera.MAUI.CameraResult> Camera.MAUI.CameraView.StartCameraAsync(Microsoft.Maui.Graphics.Size)'

System.MissingMethodException: 'Method not found: System.Threading.Tasks.Task`1<Camera.MAUI.CameraResult> Camera.MAUI.CameraView.StartCameraAsync(Microsoft.Maui.Graphics.Size)'

Tested on a Samsung a52 and Google Pixel 7 Pro.

Any help will be highly appreciated.

hjam40, 2023-05-25:
Hi Simon,

In the new version, StartCameraAsync has a parameter for set photo resolution, but It has a default value of 0 (max resolution). Your problem looks like you are using a an old and new plugging versions? I'm not sure why you are getting this error..

hjam40, 2023-05-25:
Hi Francois,

Changelog is in my to do list... :)