hjam40/Camera.MAUI #42: poor image quality

jarvis10211, 2023-05-25:
Camera view quality is very bad & blurry. How to increase image qualty or resolution?

jarvis10211, 2023-05-25:
bad image quality in barcode scan camera viewfinder at both IOS and android

Maryusz, 2023-05-26:
I use this library to scan qr codes too, and I have the same issue I suppose. 
Using the code from the documentation when a tab shows the preview of the qr code it gets really strange effect on the camera, like the camera starts with a very low and blurry resolution. 

`if (cameraView.Cameras.Count > 0)
        {
            cameraView.Camera = cameraView.Cameras.First();
            MainThread.BeginInvokeOnMainThread(async () =>
            {
                try
                {
                    if (cameraView != null && await cameraView.StartCameraAsync() == CameraResult.Success)
                    {
                        await Task.Delay(500);
                        cameraView.BarCodeOptions = new Camera.MAUI.ZXingHelper.BarcodeDecodeOptions
                        {
                            AutoRotate = true,
                            PossibleFormats = { ZXing.BarcodeFormat.QR_CODE },
                            ReadMultipleCodes = false,
                            TryHarder = true,
                            TryInverted = true
                        };

                        cameraView.BarCodeDetectionFrameRate = 10;
                        cameraView.BarCodeDetectionMaxThreads = 2;
                        cameraView.ControlBarcodeResultDuplicate = true;
                        cameraView.BarCodeDetectionEnabled = true;
                    }
                }
                catch (Exception ex)
                {
                    Debug.WriteLine(ex.Message);

                }
            });
        }`

[image]

When I however 'restart' the view, resolution it's right. 
`    protected override void OnAppearing()
    {
        base.OnAppearing();

        MainThread.BeginInvokeOnMainThread(async () =>
        {
            try
            {
                await cameraView.StartCameraAsync();
            }
            catch (Exception ex)
            {
                Debug.WriteLine(ex.Message);
            }

        });
    }`

[image]

Now I tried to select a resolution with Size(x, y) and if I remember well, the issue persist. 
The difference is that if a Resolution is selected it is better performant (if you have a 4k camera on the phone and you don't select a resolution, it becomes hot from processing the images because it analyzes images in the highest resolution.).

manuelgroo, 2023-06-02:
I have tried calling the ForceAutoFocus method but I'm getting the same poor quality in my camera, do you have any other idea to solve this?

Maryusz, 2023-06-08:
I added too the ForceAutoFocus method call, but actually the problem persists. 
It seems that at a first start, the camera takes a low resolution, if I restart the camera (in my app on appearing, the camera is restarted) it works fine.

hjam40, 2023-06-09:
Hi,

It same to be a problem in some android devices and I can't do anything to solve it. On Android I set the AutoFocus on when you start the camera (and when you restart it is the same method...)

Maryusz, 2023-07-04:
I'll try the same approach tomorrow, thank for the suggestion @Vetleskrett 👍

kennykanp, 2023-07-18:
It's strange that this happens. In my case, I'm currently working around the issue like this:

```csharp
await cameraView.StopCameraAsync();
await cameraView.StartCameraAsync();
await Task.Delay(TimeSpan.FromSeconds(3));
if (!cameraView.IsFocused)
{
    await cameraView.StopCameraAsync();
    await cameraView.StartCameraAsync(cameraView.Camera.AvailableResolutions.OrderByDescending(size => size.Width * size.Height).FirstOrDefault());
    cameraView.ForceAutoFocus();
}
```

This workaround seems to resolve the problem, but it would be great to find a more permanent solution.

Robula, 2023-08-07:
Same quality issue if starting the camera too early, I am working around this by adding `Task.Delay(1);` before I start the camera.

Without workaround:
[image]

With workaround:
[image]

Here is my code:

```csharp
public partial class CameraPage : ContentPage, IAsyncDisposable
{
    public CameraPage()
    {
        InitializeComponent();

        CameraView.CamerasLoaded += CameraView_CamerasLoaded;
        CameraView.BarcodeDetected += CameraViewOnBarcodeDetected;
        CameraView.Loaded += CameraViewOnLoaded;
    }

    private async void CameraViewOnLoaded(object? sender, EventArgs e)
    {
        CameraView.Loaded -= CameraViewOnLoaded;
        await Task.Delay(1);  // Workaround
        await CameraView.StartCameraAsync();
    }

    // Omitted
}
```

UnreachableCode, 2023-11-21:
@Vetleskrett's workaround worked for me. I don't think it's a good idea to rely on Task.Delay because you can't be sure how different devices are going to handle time related code.

mdjavedakhtar, 2024-04-10:
i am still having the same issue on android
the available resolutions are showing 0
image is too low resolution !!

DanielWooldridge1, 2024-07-01:
Has there been any updates on this issue?

MinhVu990307, 2025-03-20:
Well in my case, my code work like this

`private async void Main_Camera_CamerasLoaded(object sender, EventArgs e)
{
    try
    {
        if (Main_Camera.NumCamerasDetected > 0)
        {
            if (Main_Camera.NumMicrophonesDetected > 0)
            {
                Main_Camera.Microphone = Main_Camera.Microphones.First();
            }
            //TO DO: Select resolution for camera fit with device
            Main_Camera.Camera = Main_Camera.Cameras.First();
            Main_Camera.FlashMode = FlashMode.Disabled;
            Main_Camera.ZoomFactor = 0f;
            Main_Camera.AutoStartPreview = false;

            SetImageViewInfo();

            //Calculate the most suitable resolution
            var listSize = Main_Camera.Camera.AvailableResolutions;
            listSize.Sort((a, b) => (b.Width * b.Height).CompareTo(a.Width * a.Height));
            _fittingSize = listSize[0];

            double screenAspectRatio = _cameraWidth / _cameraHeight;
            double minAspectRatioDiff = double.MaxValue;

            foreach (var size in listSize)
            {
                double aspectRatio = (double)size.Width / size.Height;
                double aspectRatioDiff = Math.Abs(aspectRatio - screenAspectRatio);

                if (aspectRatioDiff < minAspectRatioDiff)
                {
                    _fittingSize = size;
                    minAspectRatioDiff = aspectRatioDiff;
                }
            }

            //await Task.Delay(1);
            //await Task.Run(async () =>
            //{
            await Main_Camera.StopCameraAsync();
            await Main_Camera.StartCameraAsync(_fittingSize);
            //});
        }
    }
    catch (Exception ex)
    {
        Debug.WriteLine(ex.StackTrace);
    }
}`

`private void SetImageViewInfo()
{
    try
    {
        var mainDisplayInfo = DeviceDisplay.Current.MainDisplayInfo;
        double screenWidthPixels = mainDisplayInfo.Width;
        double screenHeightPixels = mainDisplayInfo.Height;
        _density = mainDisplayInfo.Density;

        // Convert from pixel sang DIPcal
        _cameraWidth = screenWidthPixels / _density - 80;
        _cameraHeight = screenHeightPixels / _density - 40;

        //Set width and height for cameraView
        Main_Camera.WidthRequest = _cameraWidth;         //650
        Main_Camera.HeightRequest = _cameraHeight;       //320
    }
    catch (Exception ex)
    {
        Debug.WriteLine(ex.StackTrace);
    }
}`

And this use in here: 
`private void RegisterEvent()
{
    Main_Camera.CamerasLoaded += Main_Camera_CamerasLoaded;
}`

This work fine for me, and I have high resolution for camera, not the highest I think, and the picture I took have the aspect closest to the CameraView I design in my device.

This issue is major problem to me and my client and I took alot of time, so hopefully this can help someone out there need this