hjam40/Camera.MAUI #43: When trying to open CameraAsync: System.NullReferenceException: 'Object reference not set to an instance of an object.'

GabrieleViga, 2023-05-29:
```
public async Task CameraLoadAsync()
        {
            if (mainView.camera.Cameras.Count > 0)
            {
                mainView.camera.Camera = mainView.camera.Cameras.First();

                if (mainView.camera ==  null) return;

                MainThread.BeginInvokeOnMainThread(async () =>
                {
                    await mainView.camera.StopCameraAsync();
                    await mainView.camera.StartCameraAsync();
                });
            }
        }
```
When trying to StartCameraAsync it gives this error: System.NullReferenceException: 'Object reference not set to an instance of an object.'
I checked but nothing is null neither the camera or cameraView

hjam40, 2023-06-01:
Hi

In witch platform?

UkeHa, 2023-06-28:
@hjam40, happens for me once i try to go back to the previous page 

if (await cameraView.StopCameraAsync() == CameraResult.Success)
		{
			try
			{
				await Shell.Current.GoToAsync("..", new Dictionary<string, object>()
				{
					{"Barcode",barcode }
				});

			}
			catch (Exception exception)
			{
				Console.WriteLine(exception);
				throw;
			}

migueBarrera, 2023-12-19:
I have the same issue in production. On Android 11 and 12.
It doesn't happen all the time, just sometimes. 

I use this code:

[image]

[image]

@hjam40 any idea please?

archblaze, 2024-01-16:
I'm using the same stock code as @migueBarrera and see the error pretty consistently when I'm flipping between Flashmodes from Enabled to Disabled.  To make the camera recognize the change in modes (since the camera is already started), I force a StopCameraAsync() then a StartCameraAsync() (like above) immediately before I make the call to TakePhotoAsync().  I think it becomes a race condition as the TakePhotoAsync() is the call throwing the exception.  Even if I force the routines to block so each is performed in sequence, I tend to see it.  Subsequent calls to TakePhotoAsync() don't fail unless I run it through the Stop/Start/Take sequence described above again.

migueBarrera, 2024-02-02:
@hjam40  any idea about this error??

archblaze, 2024-02-03:
> @hjam40 any idea about this error??

Actually, I'm no longer getting the error because I rewrote the code to tightly control when a StartCameraAsync() and when a StopCameraAsync() is called.  I no longer force a stop and start after changing Flashmodes, but I do still perform the stop and start after changing cameras (front to back, etc.)  Additionally, I'm only deploying onto Android 13 or greater now...that's helped a lot.  Thanks for following up!

vovanb, 2024-06-03:
same problem - after call to await cameraViewMy.StopCameraAsync(); and then (await cameraViewMy.StartCameraAsync() give exception cameraViewMy.TakePhotoAsync(ImageFormat.JPEG)