hjam40/Camera.MAUI #44: QR reader wont work when CommunityToolkit.Maui is installed

mfabris1, 2023-05-29:
I'm using Camera.MAUI 1.4.3 to read QR code in my app, but if I add to my solution the nuget package [CommunityToolkit.Maui](https://www.nuget.org/packages/CommunityToolkit.Maui) it suddently stops working. Camera open correctly but it wont read the QR code anymore. Is this a known compatibilty issue?

mfabris1, 2023-05-29:
Out of curiosity I've downgraded Camera.MAUI to 1.4.2 and this way it seems working fine with CommunityToolkit.Maui

hjam40, 2023-06-01:
Hi,

Until I know, this plugging has not any incompatibility with CommunityToolkit.Maui. I didn't add nothing new from 1.4.2 to 1.4.3 for QR detection. Maybe the problem could be with Zxing package?

Hottemax, 2023-06-09:
Any new insights or developments regarding the QR code detection in the latest version?

hjam40, 2023-06-09:
Hi,

I have released a new version 1.4.4. I have test it in a test app with communitytoolkit installed on android device and it works fine for me with QR code detection. I didn't really change anything from the previews version in android code, maybe it had an error in nuget implementation....

mfabris1, 2023-06-22:
Hi, just tried out with the 1.4.4 and it seems to work with both Android and iOS. Altough i must say that in order to make it detected the QR code with Android you have to be carefull not to be too close to the QR because even if the camera focus is right it wont detect it, but as soon as you step away a lil bit from it detects it. 
I think we can consider this issue closed

closed 2023-06-22 by mfabris1