hjam40/Camera.MAUI #45: camera do not work in window

TheR7angelo, 2023-05-29:
I want to display my webcam on window, and camera on android.
In android that work but in window no why ?

My code:

`public MainPage()
    {
        InitializeComponent();
        foreach (var i in CameraView.Cameras)
        {
            Console.WriteLine(i.Name);
        }
    }

    private void CameraView_CamerasLoaded(object sender, EventArgs e)
    {
        if (CameraView.NumCamerasDetected <= 0) return;
        
        if (CameraView.NumMicrophonesDetected > 0) CameraView.Microphone = CameraView.Microphones.First();
        CameraView.Camera = CameraView.Cameras.First();
            
        MainThread.BeginInvokeOnMainThread(async () =>
        {
            await CameraView.StartCameraAsync();
        });
    }`

my Package.appxmanifest:
`  <Capabilities>
    <rescap:Capability Name="runFullTrust" />
    <DeviceCapability Name="webcam"/>
    <DeviceCapability Name="microphone"/>
  </Capabilities>`

hjam40, 2023-06-01:
Hi,

It is a known problem with Windows, you could add a Delay before the StartCameraAsync and it resolve the problem. In the next plugging version It should works fine.

TheR7angelo, 2023-06-01:
Hi,

I just added
`
Thread.Sleep(1000)
`
It doesn't change anything, is it normal?
I don't have to use the right delay maybe

hjam40, 2023-06-09:
Hi,

In the new version 1.4.4 I have added a new windows control for check if the native control is loaded before to initialize the preview, It should work fine right now (almost in my testapp works fine).

TheR7angelo, 2023-06-11:
Hi
I do not understand I followed your read me and it does not work, my application does not start.
I used your test project and it works perfectly thank you.

Can you update your read me or make it more understandable?

xufeitt, 2023-06-24:
I want to know too ,  StartCameraAsync  how to work well in windows ?