hjam40/Camera.MAUI #46: please add a library openCV to the library

AbdalaMask, 2023-05-29:
Thank you very much
Could you please add a library openCV to the library

FrKaram, 2023-05-30:
You mean shape detection right ?

AbdalaMask, 2023-05-31:
to operate the library MAUI

AbdalaMask, 2023-05-31:
to operate on the camera to detect
 openCV

AbdalaMask, 2023-05-31:
To operate on the camera to detect and use the library in Android

AbdalaMask, 2023-05-31:
In short use openCV on MAUI

hjam40, 2023-06-01:
Hi,

You can add the OpenCV package to your application and use it with the images obtained from this plugging.
For the shape, you can read the properties HorizontalViewAngle and VerticalViewAngle from the Camera property and by the moment, only in Android.

Expecho, 2023-06-09:
@hjam40 Forgive me if I ask a question I should have know the answer to but for now, it seems the only way to get an image using this plugin is by doing something like `cameraView.GetSnapShot(Camera.MAUI.ImageFormat.JPEG);` or am I wrong?

The built-in barcode/QR detection uses the `FrameReader_FrameArrived` event as seen [here](https://github.com/hjam40/Camera.MAUI/blob/d9ee42a428b9ee41efff624fe01643b62c7781e6/Camera.MAUI/Platforms/Windows/MauiCameraView.cs#L362) on the Windows platform (haven't looked at the other platform implementations). How about making this event public so consumers can subscribe and process the received frames by, for example, ML algorithms

AbdalaMask, 2023-06-09:
hjam40
Is there an example to add  the OpenCV library to Camera.MAUI

Expecho, 2023-06-10:
> Hi,
> 
> You can add the OpenCV package to your application and use it **with the images obtained from this plugging**. For the shape, you can read the properties HorizontalViewAngle and VerticalViewAngle from the Camera property and by the moment, only in Android.

AFAIK there are limited options to get access the the images obtained from this plugin. I've opened #53 to start a discussion for this.