hjam40/Camera.MAUI #50: CameraView takes over iOS popup

Jensevent, 2023-06-07:
When using the Popups from the Maui community toolkit on iOS, the cameraView takes over all elements, making it impossible to use buttons.

Steps to reproduce:
1. Create new Maui Blazor app
2. Install Camera.MAUI en Communitytoolkit.Maui
3. Create a Community toolkit popup and attach a popup using the following method (https://youtu.be/2dllz4NZC0I?t=918, 3rd option)
4. Add a button to the popup with the function Close()
5. Run the app

https://github.com/hjam40/Camera.MAUI/assets/35815436/975f30d7-baa2-42b5-ac67-e4e6d7ce5a6d

On android this will work perfectly fine, but when deploying on iOS the button can not be pressed.
The code for this problem can also be found here (https://github.com/Jensevent/PopUpTest). In this app myPopup is my Community toolkit popup.
I am trying to build to an Iphone 7, working from a mac mini with Ventura 13.4. If you need anything more, just ask.

hjam40, 2023-06-09:
Hi,

Sorry, but I can't change the community toolkit pop up behavior. On iOS this control is a UIView with no special requirements, maybe you could set a required width and height for the control...

ToastinYou, 2023-10-30:
I am having a similar experience as the OP. Except, I am not using any sort of Popup. Simply a page with a "Start" camera button, the camera opens and fills the page, and I have added a "Cancel" button over the camera view. The button is not clickable when the camera is active.

`        <AbsoluteLayout>
            <Button Text="Start" AbsoluteLayout.LayoutBounds="0, 1, 1, 0.5" AbsoluteLayout.LayoutFlags="All" BackgroundColor="Brown" Command="{Binding StartCamera}" />
            <cv:CameraView x:Name="cameraView" IsVisible="{Binding AutoStartPreview}" AbsoluteLayout.LayoutBounds="0, 0, 1, 1" AbsoluteLayout.LayoutFlags="All" BarCodeDetectionEnabled="True" BarCodeOptions="{Binding BarCodeOptions}" BarCodeResults="{Binding BarCodeResults, Mode=OneWayToSource}"
                Cameras="{Binding Cameras, Mode=OneWayToSource}" Camera="{Binding Camera}" AutoStartPreview="{Binding AutoStartPreview}" NumCamerasDetected="{Binding NumCameras, Mode=OneWayToSource}" />
            <Button Text="Cancel" IsVisible="{Binding AutoStartPreview}" Command="{Binding StopCamera}" AbsoluteLayout.LayoutBounds="0, 0, 0.5, 0.5" AbsoluteLayout.LayoutFlags="All" BackgroundColor="Black" TextColor="White" Opacity="1.0" CornerRadius="0" />
        </AbsoluteLayout>`