hjam40/Camera.MAUI #53: Providing access to idividual frames.

Expecho, 2023-06-10:
For now, it seems the only way to get an image using this plugin is by doing something like `cameraView.GetSnapShot(Camera.MAUI.ImageFormat.JPEG);` or am I wrong?

The built-in barcode/QR detection uses the `FrameReader_FrameArrived` event as seen [here](https://github.com/hjam40/Camera.MAUI/blob/d9ee42a428b9ee41efff624fe01643b62c7781e6/Camera.MAUI/Platforms/Windows/MauiCameraView.cs#L362) on the Windows platform (haven't looked at the other platform implementations). How about making this event public so consumers can subscribe and process the received frames by, for example, ML algorithms, other QR/Barcode libs or whatever one needs?

If it fits your purpose I am more than willing to try to come up with an implementation, although testing on an IOS device would be hard for me as I don't have one.

cropyai, 2023-06-13:
@Expecho We also wanted to get individual frames (5 per second). However, the GetSnapShot method works so slowly that we can only get 1 per second. If we could access the photo-capturing event, it would be awesome.

hjam40, 2023-06-21:
Hi,

I understand your problem, but you have to think that this plugging is not an Android Camera Preview, it is a multiplatform plugging. This is a problem because each platform gets a different image format from the camera frame and I have to convert it to a common format (this is the reason why GetSnapShot is so slowly). By the way, right now you could try to use the auto snap properties (in memory buffers):

AutoSnapShotSeconds = 0.2;
AutoSnapShotFormat = ImageFormat.JPG;
AutoSnapShotAsImageSource = false;

and read from the bindable property SnapShotStream.

Hrishikesh46779, 2023-09-04:
It will be very helpful if we get access to the raw live camera feed and access each frame.

kfrancis, 2023-11-20:
Agree, doing that means we can run OCR on the frame data making it more than just barcodes.

cropyai, 2023-11-20:
@kfrancis @Hrishikesh46779 I can maybe help you in accessing live preview frames? I kind of edited the library.

programatix, 2023-11-22:
I've made some modification by moving qr code decoding as plugin. See https://github.com/hjam40/Camera.MAUI/pull/107. You can create your own plugin to process the frame provided by Camera.MAUI.