hjam40/Camera.MAUI #57: No fullscreen barcode detection on Android

InkFeather, 2023-06-17:
Hi there,

I use the CameraView to detect barcodes on Android. Everything works pretty fine as long as I set fixed dimensions on the view: 

```
<cv:CameraView
        x:Name="CameraComponent"
        WidthRequest="450"
        HeightRequest="300"
        BarCodeDetectionEnabled="True"
        BarCodeDetectionFrameRate="10"
        BarCodeDetectionMaxThreads="5"
        CamerasLoaded="CameraView_CamerasLoaded"
        BarcodeDetected="CameraComponent_BarcodeDetected"/>
```

Things go wrong when I try to expand the view to full screen:

```
<cv:CameraView
        x:Name="CameraComponent"
        HorizontalOptions="FillAndExpand"
        VerticalOptions="FillAndExpand"
        BarCodeDetectionEnabled="True"
        BarCodeDetectionFrameRate="10"
        BarCodeDetectionMaxThreads="5"
        CamerasLoaded="CameraView_CamerasLoaded"
        BarcodeDetected="CameraComponent_BarcodeDetected"/>
```

Barcode detection doesn't trigger anymore. I also tried to encapsulate the view into a grid with the same result.

Did I missed something ?

adrianotrentim, 2023-08-04:
Same problem!!!!

hiepis, 2023-08-08:
same problem - version 1.4.4

Rajnikant-Civica, 2023-08-25:
same problem - version 1.4.4

HaydenFerries, 2023-08-28:
Same issue here

UkeHa, 2023-08-28:
Binding WidthRequest/Heightrequest and setting via DeviceDisplay.MainDisplayInfo.Height in the constructor also doesn't solve it. Only specific height/width works.

hiepis, 2023-08-30:
Fixed: 
MauiCameraView-Android
- Remove timer
AddTarget at StartPreview()
 - previewBuilder.AddTarget(imgReader.Surface);
Decode at ImageAvailableListener
  image.Close(); 
  Bitmap bitmap = BitmapFactory.DecodeByteArray(imageData, 0, imageData.Length);
  cameraView.DecodeBarcode(bitmap);

UkeHa, 2023-08-30:
@hiepis could you provide more info/examples?

hiepis, 2023-08-31:
[Camera.MAUI-master.zip](https://github.com/hjam40/Camera.MAUI/files/12484097/Camera.MAUI-master.zip)
@UkeHa  Edit on this project - menu fullscreen

UkeHa, 2023-08-31:
@hiepis, could you open a pull request, so that @hjam40 can merge it?

WilliamWatterson86, 2023-10-03:
Not sure if anyone found a solution but I was able to get mine working with the code below:

In xaml define a Stacklayout filling the page:

`<StackLayout IgnoreSafeArea="True" AutomationProperties.IsInAccessibleTree="False" x:Name="stkScannerView" Grid.RowSpan="2" HorizontalOptions="FillAndExpand" VerticalOptions="FillAndExpand" Padding="0" Spacing="0"></StackLayout>`

In cs file:

```
public ScanQrCodePage(ScanQrCodePageViewModel vm)
{
	InitializeComponent();
    this.BindingContext = vm;

    vm.RefocusScanningAction = this.RefocusCamera;
    vm.ToggleTorchAction = this.ToggleTorch;

    this.cameraView = new CameraView();

    this.cameraView.CamerasLoaded += this.cameraView_CamerasLoaded;
    cameraView.BarCodeDetectionFrameRate = 5;
    cameraView.BarCodeDetectionMaxThreads = 5;

    cameraView.BarCodeOptions = new Camera.MAUI.ZXingHelper.BarcodeDecodeOptions
    {
        PossibleFormats = { ZXing.BarcodeFormat.QR_CODE },
        ReadMultipleCodes = false,
        TryHarder = true,
        AutoRotate = false,
    };
}

private void ExecuteStartScanning()
{
    try
    {
        this.cameraView.CamerasLoaded -= this.cameraView_CamerasLoaded;
        this.cameraView.CamerasLoaded += this.cameraView_CamerasLoaded;
        this.cameraView.BarcodeDetected -= this.cameraView_BarcodeDetected;
        this.cameraView.BarcodeDetected += this.cameraView_BarcodeDetected;

        MainThread.BeginInvokeOnMainThread(async () =>
        {
            try
            {
                await Task.Delay(150);

                if (!this.stkScannerView.Children.Any())
                {
                    this.stkScannerView.Children.Add(this.cameraView);
                }

                await cameraView.StartCameraAsync();
            }
            catch (Exception ex)
            {
                Logger.LogException(ex, this);
            }
        });
    }
    catch (Exception ex)
    {
        Logger.LogException(ex, this);
    }
}

private void ExecuteStopScanning()
{
    try
    {
        this.cameraView.CamerasLoaded -= this.cameraView_CamerasLoaded;
        this.cameraView.BarcodeDetected -= this.cameraView_BarcodeDetected;
        this.cameraView.BarCodeDetectionEnabled = false;

        MainThread.BeginInvokeOnMainThread(async () =>
        {
            try
            {
                await cameraView.StopCameraAsync();

                if (this.cameraView.TorchEnabled)
                {
                    this.cameraView.TorchEnabled = false;
                    this.frmToggleTorch.Opacity = 0.5;
                }

                if (this.stkScannerView.Children.Any())
                {
                    this.stkScannerView.Children.Remove(this.cameraView);
                }

            }
            catch (Exception ex)
            {
                Logger.LogException(ex, this);
            }
        });
    }
    catch (Exception ex)
    {
        Logger.LogException(ex, this);
    }
}

private void cameraView_BarcodeDetected(object sender, Camera.MAUI.ZXingHelper.BarcodeEventArgs args)
{
    if (args != null && args.Result.Any())
    {
        MainThread.BeginInvokeOnMainThread(() =>
        {
            try
            {
                if (this.BindingContext is ScanQrCodePageViewModel vm)
                {
                    vm.ScanCommand?.Execute(args);
                }
            }
            catch (Exception ex)
            {
                Logger.LogException(ex, this);
            }
        });
    }
}

private void cameraView_CamerasLoaded(object sender, EventArgs e)
{
    if (cameraView.Cameras.Count > 0)
    {
        cameraView.Camera = cameraView.Cameras.First();
    }
}
```

WoodyJ007, 2023-10-09:
This still appears to be an issue.  Is there a simple fix?

UkeHa, 2023-10-09:
@WoodyJ007, yes, it's called using this https://github.com/JimmyPun610/BarcodeScanner.Mobile project instead. Everything works as intended and is blazing fast!