hjam40/Camera.MAUI #58: AvailableResolutions for Windows shows zero; max size 640 x 480

ClintMayers, 2023-06-19:
List<Size> sizes = cameraView.Camera.AvailableResolutions; // returns empty on Win.
- using Logitech Brio 4K camera 3840*2160 is max size.

When passing in Size for either for StartCameraAsync or StartRecordingAsync:
var result = await cameraView.StartCameraAsync(new Size(1280,720)); // for example...
- always returns: CameraResult.ResolutionNotAvailable even when assigning 640x480.
- However, will record 640x480 when left empty (default) or assigning new Size(0,0).

Works as expected on Android.

So far, MacOS (M1) does list Sizes but only 4 options with max size of 1920 x 1080.  (even with the exact same web camera, but is likely a driver issue and not a big deal for me at the moment).  I am still playing with the permissions on macOS to record.

Since we cannot use Cameras in macOS's Simulator for iOS, and I don't have access to an iPhone at the moment, I'll have to wait to test that.

Conclusion: Windows Camera Setting and Info is not yet incorporated--at least not for Logitec Brio 4K 30p camera I have.

hjam40, 2023-06-21:
Hi Client,

I don't know what is the problem with this camera. Actually I'm reading the available resolutions from the camera profile:
`
                    foreach(var profile in MediaCapture.FindAllVideoProfiles(s.Id))
                    {
                        foreach (var recordMediaP in profile.SupportedRecordMediaDescription)
`
In my case from my laptop integrated camera I get 9 different resolutions from 1280x720 to 320x180. In my other windows computer with an usb camera I get a similar result. Maybe is a local problem with the driver but I only can read what windows shows me...

ClintMayers, 2023-06-21:
Thanks, I'll do some troubleshooting on my end, and share what I can.