hjam40/Camera.MAUI #78: Performance on iOS with 'PossibleFormats'

marco-skizza, 2023-07-27:
Hi

I set up a new project and used `CameraView` with `BarCodeDetectionEnabled="True"`.
In the code behind I set the `PossibleFormats`:
```
        CameraView.BarCodeOptions = new BarcodeDecodeOptions
        {
            PossibleFormats = { ZXing.BarcodeFormat.CODE_128 }
        };
```
When run on an iPad mini 6, after a while the app gets unresponsive and the iPad gets hot.
So I guess there must be some high CPU consumption.

Full code `MainPage.xaml`:
```
<?xml version="1.0" encoding="utf-8" ?>
<ContentPage xmlns="http://schemas.microsoft.com/dotnet/2021/maui"
             xmlns:x="http://schemas.microsoft.com/winfx/2009/xaml"
             xmlns:cv="clr-namespace:Camera.MAUI;assembly=Camera.MAUI"
             x:Class="MauiCamera.MainPage">

    <ScrollView>
        <Grid>
            <Grid.RowDefinitions>
                <RowDefinition Height="*"/>
                <RowDefinition Height="50"/>
            </Grid.RowDefinitions>

            <cv:CameraView
                Grid.Row="0"
                x:Name="CameraView"
                BarCodeDetectionEnabled="True"
                CamerasLoaded="CameraView_OnCamerasLoaded"
                BarcodeDetected="CameraView_OnBarcodeDetected" />

            <Label
                Grid.Row="1"
                x:Name="Text"
                HorizontalOptions="Center" />

        </Grid>
    </ScrollView>

</ContentPage>
```
Full code `MainPage.xaml.cs`:
```
using Camera.MAUI.ZXingHelper;

namespace MauiCamera;

public partial class MainPage : ContentPage
{
    private int _count;

    public MainPage()
    {
        InitializeComponent();
        Text.Text = "-";
    }

    private async void CameraView_OnCamerasLoaded(object sender, EventArgs e)
    {
        await Task.Delay(3000);

        if (CameraView.NumCamerasDetected == 0)
        {
            return;
        }

        CameraView.BarCodeOptions = new BarcodeDecodeOptions
        {
            PossibleFormats = { ZXing.BarcodeFormat.CODE_128 }
        };

        CameraView.Camera = CameraView.Cameras.First();
            
        MainThread.BeginInvokeOnMainThread(async () =>
        {
            await CameraView.StartCameraAsync();
        });
    }

    private void CameraView_OnBarcodeDetected(object sender, BarcodeEventArgs args)
    {
        _count++;
        Text.Text = $"{_count}: {args.Result.First().Text}";
    }
}
```

garcia252, 2024-05-04:
Did you solve this @marco-skizza ? Experiencing similar issue on our iPad with respect to unresponsiveness.

marco-skizza, 2024-05-05:
Hi @garcia252
Back then, I just didn't set the `BarCodeOptions` and let it scan for all possible barcode formats.
But in the meantime I switched to another library, because I wasn't satisfied with the barcode detection quality...