hjam40/Camera.MAUI #94: [iOS] MVVM - Binding BarCodeDetectionEnabled makes page unresponsive

UkeHa, 2023-08-28:
Binding BarCodeDetectionEnabled makes the entire page unresponsive. 
Neither buttons nor the back swipe/button to navigate from the page stop reacting. I've bound an EventToCommandBehavior to a detected barcode. Scanning a barcode is the only way to leave the page.

Steps to reproduce:

- Use MVVM Sample
- Add Binding for BarCodeDetectionEnabled and set it via ApplyQueryAttributes on page
- Open Camera-Page and try to go back

UkeHa, 2023-08-29:
@hjam40, this happens in both .net7 as well as the latest .net8 preview of MAUI.

Felicity-R, 2023-09-28:
I have the same problem. I also noticed that if I tried to tap on anything while the camera was up, those taps would finally get processed once the camera was closed (i.e. after I scanned a barcode). So after the camera was dismissed my app would be bombarded with queued-up tap gestures. So it's not that the input isn't getting through at all, it's more like the camera blocks the UI thread so nothing else has a chance to get processed while it's running.

UkeHa, 2023-09-28:
@Felicity-R, honestly, just switch to this https://github.com/afriscic/BarcodeScanning.Native.Maui and never look back. It's **hands down the best** way to do camera/barcode stuff. It just works and is so freaking fast!

Felicity-R, 2023-09-28:
> @Felicity-R, honestly, just switch to this https://github.com/afriscic/BarcodeScanning.Native.Maui and never look back. It's **hands down the best** way to do camera/barcode stuff. It just works and is so freaking fast!

Thanks for the recommendation! Unfortunately I also need Windows support, but maybe I'll check that out for iOS.

UkeHa, 2023-09-28:
then i'd recommend to use camera.maui for windows and do the rest over the one above. It truly is THAT much better.