hjam40/Camera.MAUI #97: [Error] QR code not detected

Kiriox94, 2023-09-11:
Hello I set up this project in my application to make a QR code scanner and at the beginning when I tested it worked perfectly but after numerous modifications in my application the QR codes are not detected.
Here is the code of the page with the qr code reader. Please help me
(I should point out that I can see the video from my webcam clearly)

QrCodeScannerPage.xaml:
```c#
<?xml version="1.0" encoding="utf-8" ?>
<ContentPage xmlns="http://schemas.microsoft.com/dotnet/2021/maui"
             xmlns:x="http://schemas.microsoft.com/winfx/2009/xaml"
             x:Class="MauiApp1.QrCodeScannerPage"
             xmlns:cv="clr-namespace:Camera.MAUI;assembly=Camera.MAUI"
             Title="Scannez un Qr code">
    <Grid>
        <Label x:Name="invalidQr" 
               TextColor="Red" 
               Text="Qr code invalide"
               HorizontalOptions="Center" 
               VerticalOptions="Center"
               ZIndex="3" 
               Grid.Row="0" 
               Margin="0,0,0,450" 
               FontSize="Large"
               IsVisible="False"/>
        <Image Source="qr_scanner_border.png" WidthRequest="300" HeightRequest="300" HorizontalOptions="Center" ZIndex="2" Grid.Row="0" />
        <cv:CameraView x:Name="cameraView" HorizontalOptions="FillAndExpand" VerticalOptions="FillAndExpand" ZIndex="1" Grid.Row="0" />
    </Grid>
</ContentPage>
```

QrCodeScannerPage.xaml.cs:
```c#
using System.Text.RegularExpressions;
using Camera.MAUI;
using System.Diagnostics;

namespace MauiApp1;

public partial class QrCodeScannerPage : ContentPage
{
	public QrCodeScannerPage()
	{
		InitializeComponent();
        cameraView.CamerasLoaded += CameraView_CamerasLoaded;
        cameraView.BarcodeDetected += CameraView_BarcodeDetected;

        cameraView.BarCodeOptions = new Camera.MAUI.ZXingHelper.BarcodeDecodeOptions
        {
            AutoRotate = true,
            PossibleFormats = { ZXing.BarcodeFormat.QR_CODE },
            ReadMultipleCodes = false,
            TryHarder = true,
            TryInverted = true
        };
        cameraView.BarCodeDetectionFrameRate = 10;
        cameraView.BarCodeDetectionMaxThreads = 5;
        cameraView.ControlBarcodeResultDuplicate = true;
        cameraView.BarCodeDetectionEnabled = true;
    }

    private void CameraView_CamerasLoaded(object sender, EventArgs e)
    {
        if (cameraView.NumCamerasDetected > 0)
        {
            if (cameraView.NumMicrophonesDetected > 0)
                cameraView.Microphone = cameraView.Microphones.First();
            cameraView.Camera = cameraView.Cameras.First();
            MainThread.BeginInvokeOnMainThread(async () =>
            {
                if (await cameraView.StartCameraAsync() == CameraResult.Success)
                {
                    Debug.WriteLine("Camera started");
                }
            });
        }
    }

    private void CameraView_BarcodeDetected(object sender, Camera.MAUI.ZXingHelper.BarcodeEventArgs args)
    {
        invalidQr.Text = "Works !!!!!!!!!";
        invalidQr.IsVisible = true;
        Debug.WriteLine("Please works");

        //string pattern = @"^.+@@(\d{6})@@$";

        //Regex regex = new(pattern);
        //System.Text.RegularExpressions.Match match = regex.Match(args.Result[0].Text);

        //if (match.Success)
        //{
        //    invalidQr.IsVisible = false;
        //    Debug.WriteLine("Code is : " + match.Groups[1].Value);
        //}
        //else
        //{
        //    invalidQr.IsVisible = true;
        //}
    }
}
```

ignatiucnetinfo, 2023-11-06:
The ZXing released a new control. It just works.

Unimbecille, 2024-02-27:
Try to remove the Tryharder. I had the same issue with that.

samjuren, 2024-10-08:
Did this solve anything? My app is having the same problem

lanosgnatiuc, 2024-10-08:
@samjuren
Use ZXing for QR. This lib is of very poor quality.

samjuren, 2024-10-08:
@lanosgnatiuc
I'm having problems with ZXing. I implemented it in the project, it worked perfectly, but now as soon as the screen opens the app closes. I had to switch to Camera.MAUI, but I found problems with it too. Looking at some forums I saw that for Camera.MAUI to work you have to leave the Nuget version 1.3.5, and it worked.