hjam40/Camera.MAUI #98: Snapshot result does not fit preview

KrizCologne, 2023-09-19:
The feed shown in the CameraView is different to the resulting Snapshot file or ImageSource:

[image]

As you can see, the result-image shows more from the top and less from the bottom. I'm playing around with aspect property of the image but the problem might be that the camerafeed just showing a cutout. 

myView:
```

<VerticalStackLayout VerticalOptions="FillAndExpand" HorizontalOptions="FillAndExpand">
        <cv:CameraView x:Name="cameraView" WidthRequest="300" HeightRequest="300" CamerasLoaded="CameraView_CamerasLoaded" AutoSnapShotFormat="PNG"/>
        <Button Text="Smile" Clicked="Button_Clicked"/>
        <Image x:Name="myIMage" WidthRequest="300" HeightRequest="300" Clip="ToBounds"/>
    </VerticalStackLayout>
```

Code behind:
```
    private async void CameraView_CamerasLoaded(object sender, EventArgs e)
    {
        cameraView.Camera = cameraView.Cameras.First();

        if (cameraView.NumCamerasDetected > 0)
        {
     
            MainThread.BeginInvokeOnMainThread(async () =>
            {
                await cameraView.StopCameraAsync();
                await cameraView.StartCameraAsync();
            });
            
        }
    }
```

```
    private async void Button_Clicked(object sender, EventArgs e)
    {        
        myIMage.Source = cameraView.GetSnapShot();
    }
```

jsgg25, 2023-11-02:
I had the same issue, the capture didn't match the preview. While reviewing the plugin code, I found something that, in my case, fixed the error
[image]

In this line, a 0 was being passed for the x and y positions, which represent the starting point of the bitmap. However, if the preview does not match the maximum screen camera size, that offset must be considered.

` bitmap = Bitmap.CreateBitmap(bitmap, (bitmap.Width - Width)/2, (bitmap.Height - Height) / 2, Width, Height);
`

KrizCologne, 2023-11-03:
Great, that did the job! @jsgg25 Thank you!

closed 2023-11-03 by KrizCologne

aleksandervalle, 2024-02-01:
This does indeed fix the issue but this change is not done in the repo, and so the issue should still be open.

janusw, 2024-02-22:
FYI: This is fixed in release [1.4.7](https://www.nuget.org/packages/CameraMaui/1.4.7), built from [my fork](https://github.com/janusw/Camera.Maui).

yassirtl, 2024-03-14:
How 

> I had the same issue, the capture didn't match the preview. While reviewing the plugin code, I found something that, in my case, fixed the error [...]

How could i use this bitmap function to solve the problem ?