hjam40/Camera.MAUI #100: .NET MAUI barcode scanner does not work for 1D codes

lukasz-strus, 2023-10-08:
I have a problem with scanning codes using the Camera.MAUI library. When I scan QR codes, everything works as it should. When I want to scan a 1D code, the scanner does not detect it. I tried various barcodes from the Internet and from products (e.g. UHT milk barcode). It doesn't detect this code every time.

I create applications in .net 7.0. Below is the code.

I posted the code on StackOverflow:
https://stackoverflow.com/questions/77253145/net-maui-barcode-scanner-does-not-work-for-1d-codes

JAVersteeg, 2023-10-17:
I have the exact same problem

Uridel, 2024-03-18:
I didn't have this problem with previous versions. However it has appeared in the latest version see #145 Did you ever find a workaround?

nmg196, 2024-08-06:
Same problem here.  QR codes are fine by my my iPhone will not scan any 1D barcodes with the current version of this library.