hjam40/Camera.MAUI #102: App crash on selecting a different camera

Nobody84, 2023-10-13:
When I try to select a different camera, the app crashes in approx. 1 from 4 tries. It semes not to be camera specific.
I can reproduce the crash with the Camera.MAUI.Test app.

Ich get the following error (complete crash log attached):
> JNI DETECTED ERROR IN APPLICATION: JNI ERROR (app bug): jobject is an invalid global reference: 0xab4a (deleted reference at index 1370)

[appcrashoncamselect_crashlog.txt](https://github.com/hjam40/Camera.MAUI/files/12890300/appcrashoncamselect_crashlog.txt)
[appcrashoncamselect_completelog.txt](https://github.com/hjam40/Camera.MAUI/files/12890301/appcrashoncamselect_completelog.txt)

**System**:
Google Pixel 6A
Android 13

jspuij, 2023-10-15:
Same issue here.

programatix, 2023-12-04:
You can try my pull (https://github.com/hjam40/Camera.MAUI/pull/107). It should have resolved almost all the crashes.