hjam40/Camera.MAUI #104: Video Recording is not worrking properly

PrathamGheewala, 2023-10-26:
Hi @hjam40 you have created really a great Nuget Package. I appreciate your work but need to work on video recording.

Video Recording is working properly. Video is not visible only audio is working.

https://github.com/hjam40/Camera.MAUI/assets/91669343/bf0cb388-d6c5-4a1b-b950-304e9a4b8952

ignatiucnetinfo, 2023-11-06:
This control is bugged as hell. I've tried to fix the code - it's a mess. Try find something else.

tpgalan, 2023-12-13:
@ignatiucnetinfo @PrathamGheewala did you manage to solve this or use another library? 
@PrathamGheewala would you be able to share your code on the start record / stop record?

ssianky, 2023-12-14:
> did you manage to solve this or use another library?

@tpgalan Use something else. This lib's code is of very bad quality.

ignatiucnetinfo, 2023-12-14:
@tpgalan 
I've used the ZXing.Net.Maui.Controls for QR reading. Works perfectly.

janusw, 2024-01-28:
You don't say on which OS (and which version) you observe the problem, but some recording issues on iOS and Android were fixed on [my fork](https://github.com/janusw/Camera.Maui) in release [1.4.6.](https://www.nuget.org/packages/CameraMaui/1.4.6)

PrathamsGithub, 2024-01-29:
It was Android 11 and Version was 1.4.4

janusw, 2024-01-29:
> It was Android 11 and Version was 1.4.4

Thanks for the info. Does release 1.4.6 (which I built yesterday) work for you?

PrathamsGithub, 2024-01-29:
I haven't try that. Will it work for .net 7 ? Because I am using Mac

janusw, 2024-01-29:
> I haven't try that. Will it work for .net 7 ?

Absolutely. Version 1.4.6 has net7.0 and net8.0 target frameworks.

> Because I am using Mac

That shouldn't be a problem. You can even build net8.0 targets on Mac.

Could you try the package and report if it works for you?