hjam40/Camera.MAUI #113: zero byte video file on ios 17

apoland, 2023-11-16:
Running the Camera.MAUI.Test test app on iOS 17 on an iPhone 15 Pro the video file seems to be corrupted (it produces a file that is zero bytes long) after recording.  This worked last time I tried it on iOS 16.

I added a couple of lines to debug:
https://github.com/hjam40/Camera.MAUI/compare/master...apoland:Camera.MAUI:debug-video?expand=1#diff-a5e8cccef5d51b0fc414bd64a4e84c580535f101b756ffac87877dfed7a114c1R106-R107

```
[0:] Start recording result Success
Thread started:  #9
[0:] Stop recording result Success
Thread started:  #10
[0:] Video size: 0
Thread started:  #11
[0:] MediaManager: Error: Cannot Open - This media may be damaged.
```

shahidjameelbhatti, 2023-12-16:
i am facing the same problem of zero size video file, is anyone able to fix this?

shahidjameelbhatti, 2023-12-16:
> Running the Camera.MAUI.Test test app on iOS 17 on an iPhone 15 Pro the video file seems to be corrupted (it produces a file that is zero bytes long) after recording. This worked last time I tried [...]

did you found any solution to it?

rdurish, 2024-01-10:
I have the same issue as well.

janusw, 2024-01-28:
FYI, this issue is fixed on [my fork](https://github.com/janusw/Camera.Maui) in [release 1.4.6](https://www.nuget.org/packages/CameraMaui/1.4.6).