hjam40/Camera.MAUI #115: access to usb-cam

reschresch, 2023-11-23:
Hello,
I have an external usb-cam connected to my Android-Phone.
When I have a look at the avaiable devices with "cameraView.Cameras" I get only the internal camaras (back + front).

How can I connect it to the USB-Cam?

Best regards,

resch

vrho94, 2023-11-29:
Hi i also need this to access my external USB camera. Did you get any solutions?

reschresch, 2023-11-29:
Hello, currently I got no answer - and no solution. 
I can't believe no one else has this problem to connecting an external cam.

vrho94, 2023-11-30:
Do you have any idea for other platforms and frameworks like for Java,Python...

hjam40, 2024-02-29:
Hi,
In this library I'm using CameraManager GetCameraIdList for get all device cameras. If you could get usb cameras, maybe it is an application usb access permission error or an android internal error. Sorry, but I can not do nothing to fix it.

closed 2024-02-29 by hjam40

vovanb, 2025-08-04:
any idea how to connect to USB camera?