hjam40/Camera.MAUI #118: CameraView in .net MAUI taking too much time for scanning barcode and QR code

vplife1, 2023-11-29:
CameraView Performance Issue: Slow Barcode/QR Code Scanning

I am experiencing a significant performance issue with the CameraView in my .NET MAUI app when scanning barcodes and QR codes. The scanning process is taking much longer than expected, leading to delays in my application.

UI Code
<Label x:Name="barcodeResult" FontSize="20"/> <cv:CameraView x:Name="cameraView" WidthRequest="300" HeightRequest="200" CamerasLoaded="cameraView_CamerasLoaded" BarCodeDetectionEnabled="True" BarcodeDetected="cameraView_BarcodeDetected"/>

Scan code
`private void cameraView_CamerasLoaded(object sender, EventArgs e)
{
if (cameraView.Cameras.Count > 0)
{
cameraView.Camera = cameraView.Cameras.First();
MainThread.BeginInvokeOnMainThread(async () =>
{
await cameraView.StopCameraAsync();
await cameraView.StartCameraAsync();
});
}
}

    private void cameraView_BarcodeDetected(object sender, Camera.MAUI.ZXingHelper.BarcodeEventArgs args)
    {
        MainThread.BeginInvokeOnMainThread(() =>
        {
            barcodeResult.Text = $"{args.Result[0].BarcodeFormat}: {args.Result[0].Text}";
        });
    }`
    
    
    

Steps to Reproduce
Launch the .NET MAUI app.
Navigate to the page containing the CameraView for barcode/QR code scanning.
Attempt to scan a barcode or QR code.
Observe the time it takes for the scanning process to complete.

Expected Behavior:
I expect the barcode/QR code scanning process to be quick and responsive, providing results promptly.

Actual Behavior:
The scanning process is noticeably slow, and there is a significant delay before the results are obtained.

sechabamot, 2023-12-13:
Are you building for Android?

vplife1, 2023-12-13:
@sechabamot  Yes

vplife1, 2023-12-18:
@programatix  @sechabamot Anything found ?

programatix, 2023-12-18:
> @programatix @sechabamot Anything found ?

As mentioned, DEBUG build is indeed very slow. Try RELEASE build. For me, RELEASE build performs a lot better, about the same as other barcode scanning projects.

garcia252, 2024-05-02:
Did you find a solution to this @vplife1 ? We've been trying the latest version of 1.5.1 and it seems very slow. 1.2.1 looks better but how did you solve this?

thomasgalliker, 2024-09-13:
Are you using the ZXing implementation? This is an issue since the first appearance of zxing. We used to suffer with this in our Xamarin app. We now use Android/iOS native barcode scanning capabilities (e.g. CameraScanner.Maui, BarcodeScanning.Native.Maui).