hjam40/Camera.MAUI #122: Not recording video on IOS 17

shahidjameelbhatti, 2023-12-16:
Hello, i am using this plugin to record video in my app, it was working perfectly before IOS 17, but now in IOS 17 its not recording the video, and saves an 0 bytes file, not a signle byte is written in that,
Can anyone help me fixing it for IOS 17?
Its very URGENT for me.
Thanks.

Luca-21, 2024-01-03:
I have the same problem with android, did u find a solution?

rdurish, 2024-01-10:
Same issue here, seems to work for android though.....

harima34, 2024-01-14:
Same for me, did you solve it ?

rdurish, 2024-01-16:
Completely stuck, iOS 17 with net8.0 saves as 0 byes with a successful result return.  @hjam40 I noticed a few changes yesterday?  Will you be pushing those up to nuget?

blago83, 2024-01-17:
In MauiCameraView.cs -> StartRecordingAsync method remove the lines to create file if it doesn't exists. Delete the file if already exists. Now it's recoding video.

[image]

janusw, 2024-01-28:
FYI, this issue is fixed on [my fork](https://github.com/janusw/Camera.Maui) in [release 1.4.6](https://www.nuget.org/packages/CameraMaui/1.4.6).

closed 2024-02-29 by hjam40