hjam40/Camera.MAUI #123: Status of this project

janusw, 2023-12-18:
Hi @hjam40,

I'd like to know what the status of this project is. There haven't been any updates for half a year and six PRs are waiting to be merged. Do you plan to continue maintaining this project, or is it dead?

It's a great project, and seems to be useful to many people. Would be a shame if it died. If you have little time for taking care of it, maybe one could share maintainership among a small group of people or so? What are your thoughts?

Best regards,
Janus

LeoJHarris, 2023-12-19:
Same here, I saw some tickets on the MAUI community project making reference to this project, if so is this going to be merged on that repo, we like many are hesitant to begin implementing a library if ceases to be supported.

janusw, 2023-12-19:
> Same here, I saw some tickets on the MAUI community project making reference to this project

you mean this one? https://github.com/CommunityToolkit/Maui/issues/259

> if so is this going to be merged on that repo, we like many are hesitant to begin implementing a library if it ceases to be supported.

I'm not sure if this is going to happen. See this PR:
https://github.com/CommunityToolkit/Maui/pull/1387
And in particular this comment:
https://github.com/CommunityToolkit/Maui/pull/1387#issuecomment-1733803425
Sounds more like CommunityToolkit will be doing its own thing.

janusw, 2023-12-19:
> Hi all, I’ve haven a lot of work this months and I couldn’t work on this project for a while, but I will be working on it for the Christmas holidays

That's good to know. Thanks for the comment, Hector. 👍 

I'll close this issue then ...

janusw, 2024-01-03:
I guess I'll reopen this issue until there is any visible progress :/

If this project continues to be unmaintained I'll probably have to fork it and maintain my own fork. I'm not very keen on that, but it's pretty much the last resort for me ...

As mentioned above I'd also be willing to help out with maintenance, instead of forking (if you are willing to accept that kind of help).

janusw, 2024-01-04:
Alright, I'm forking. I have merged most open PRs into https://github.com/janusw/Camera.Maui (in particular #52, #69, #70, #79 and #121), and I will start publishing packages to nuget.org.

hjam40, 2024-01-04:
Hi Janus,

I working on split this package in Camera and serial/qr recognize independent packages. I’m adapting it to Net 8 and testing in last version of iOS and Android. (In Spain, we are still in Christmas holidays ;) )

janusw, 2024-01-04:
> I working on split this package in Camera and serial/qr recognize independent packages. I’m adapting it to Net 8 and testing in last version of iOS and Android.

Ah, so you are actually working on a new version? I was not aware of that at all (your GitHub profile does not show any sign of life for the last seven months). Why do you do all of this in private?

And what about the open PRs? Will you take care of those? Couldn't merge them already, before starting a larger refactoring? Some of them are really simple. Deferring this will only increase the likelihood of conflicts ... !?!

> (In Spain, we are still in Christmas holidays ;) )

Yeah, in Germany we also still have school holidays, in the federal state that I live in (Hesse) even until end of next week :)

janusw, 2024-01-05:
Since I really depend on having a working package urgently, I tagged a version [1.4.5](https://github.com/janusw/Camera.Maui/releases/tag/1.4.5) in my fork now and published it on nuget.org: https://www.nuget.org/packages/CameraMaui/1.4.5

I still hope you will generate a new release at some point, @hjam40.

tpgalan, 2024-02-08:
> Hi Janus,
> 
> I working on split this package in Camera and serial/qr recognize independent packages. I’m adapting it to Net 8 and testing in last version of iOS and Android. (In Spain, we are still in Christmas holidays ;) )

Dear @hjam40, do you have an update on this? There's still some issues on this brilliant library, but now it is not 100% usable as it is. Are you still working on migrating to NET8 and splitting packages?

Thanks!

hjam40, 2024-02-08:
Hi all,

as you could see in the repo, the packages are split (and others bugs fixed), but I’m having problems with apple subscription and I can’t test it on iOS. I hope resolve it the next week and publish the new packages versions on nuget.

tpgalan, 2024-02-08:
> Hi all,
> 
> as you could see in the repo, the packages are split (and others bugs fixed), but I’m having problems with apple subscription and I can’t test it on iOS. I hope resolve it the next week and publish the new packages versions on nuget.

OK, let me know if I can help you. I have an active apple subscription and iOS devices. And also I believe we also share native language, amigo.

janusw, 2024-02-08:
> Are you still working on migrating to NET8

net8.0 target frameworks are available in version [1.4.6](https://www.nuget.org/packages/CameraMaui/1.4.6#supportedframeworks-body-tab), built from my fork.

Uridel, 2024-02-26:
I'm also waiting for the new version to drop. Any luck on being able to publish the new version @hjam40 ?

janusw, 2024-02-26:
> I'm also waiting for the new version to drop.

Are you waiting for a specific feature or bugfix? I recently dropped version [1.4.7](https://www.nuget.org/packages/CameraMaui/1.4.7) with some fixes, see https://github.com/janusw/Camera.Maui/releases/tag/1.4.7.

hjam40, 2024-02-29:
Hi,

I have released 1.5 version.