hjam40/Camera.MAUI #131: Dynamically change the CameraView size

Falco94, 2024-01-26:
Hi there,

in my application, I need to make the size of the CameraView dynamic (I want a fullscreen view, with support for orientation changing). So I do not want to set HeightRequest, and WidthRequest of the control. If i dont set these values, the `AdjustAspectRatio` function may calculate wrong values, and the image is distortet or stretched (or its not fitting inside the fullscreen; like zoomed in). This happens espacially if you change the device orientation. It seems, that after changing the orientation, somtimes the CameraView calculates (`RectF viewRect = new(0, 0, Width, Height);`) with values from the old orientation. I think it refreshes  before the actual PlatformView gets resized. With appropriate breakpoints inside the AdjustAspectRatio function, it calculates and transforms correctly. Also the preview image seems to get cut instead of having some fitting logic for the corresponding apsect ratio. In my special case, I need to open the camera preview in an 4:3 aspect ratio. 

I just call

```
var size = _cameraAspectRatios.Where(x => x.Item1.Numerator == 4 && x.Item1.Denominator == 3).MaxBy(x => x.Item2);
await CameraView.StartCameraAsync(new Size(size.Item2, size.Item3));
```

to get the highest 4:3 resolution. What I expect would be a fullscreen mode, adjusting the width and height corresponding to the given AspectRatio AND the current orientation (fitting the image, whith black borders for example).

I just tested this on Android right now.

I hope you get the idea and have a solution for my problem.

michailowski, 2024-03-07:
I am facing the same issues on Android devices. Even device performance seems to be affected, as I've been trying for two days to catch this issue occurring on one device during debugging, but not present in the release build.
On another device, which was used for tests when writing code earlier, the problem occurred very rarely even during debugging.

When rotating a device with screen rotation enabled, the device is engaged in animation playback. And judging by the debugging information, the AdjustAspectRatio function in Platform.Android.MauiCameraView.cs is called sometimes before and sometimes after the width and height have been changed according to the screen position. For this reason, further calculations are not correct and the preview is displayed crookedly on the screen.

Here is an example of debugging information when an error occurs:

**Before turn, Portrait mode:**
[0:] viewSize [w: 720 h: 1457]
[0:] videoSize [w: 2560 h: 1440]

**After turn, Landscape mode:**
[0:] viewSize [w: 720 h: 1457]
[0:] videoSize [w: 2560 h: 1440]

As you can see, the dimensions of the view have not changed.
Sometimes there are cases when the values have already been changed.

I hope this helps a little more understanding of the problem @Falco94 outlined above.

ngocnnt, 2024-06-11:
Hi @michailowski, I had the same problem as you even with release build mode. The preview screen is just not stretched when I turn the phone. I'm stuck in this situation and don't know what to do. Have you solved this problem yet?