hjam40/Camera.MAUI #138: iOS back camera upside down on start

mirceamuresanse, 2024-02-16:
On iOS the app starts with the back camera vertically mirrored (upside down), by default.
Also, when using the CameraView.MirroredImage = true, the mirror is done vertically (upside down), instead of horizontally.

janusw, 2024-03-25:
> On iOS the app starts with the back camera vertically mirrored (upside down), by default.

Do you only see this in the sample app (Camera.MAUI.Test), or also in another setup? If yes, which one?

See also https://github.com/janusw/Camera.Maui/issues/21

garcia252, 2024-05-03:
Did you solve this @mirceamuresanse ? Having the same problem that the iOS camera always starts upside down in version 1.5.1 (However in version 1.4.4 it does not)

janusw, 2025-06-27:
> I can't even get the repo to build locally:
> 
> I keep getting this:
> 
> [..]
> 
> Which ... is probably not surprising given how much .Net7 is still in this repo... I think I'll need a machine with only .Net 8 or update this repo to .Net8 only to get things to build with .Net 9 SKD installed.

@jasells You can try my fork (https://github.com/janusw/CameraMaui), which has support for .NET 8 and 9 in version 1.4.10 (current master branch).

Packages published here:
https://www.nuget.org/packages/CameraMaui

jasells, 2025-06-28:
> [@jasells](https://github.com/jasells) You can try my fork (https://github.com/janusw/CameraMaui), which has support for .NET 8 and 9 in version 1.4.10 (current master branch).
> 
> Packages published here: https://www.nuget.org/packages/CameraMaui

Is this fork based on  v 1.4 of this repo, or 1.5?   Wishing i had seen this a few months ago now.

janusw, 2025-06-28:
> > [@jasells](https://github.com/jasells) You can try my fork (https://github.com/janusw/CameraMaui), which has support for .NET 8 and 9 in version 1.4.10 (current master branch).
> > Packages published here: https://www.nuget.org/packages/CameraMaui
> 
> Is this fork based on v 1.4 of this repo, or 1.5?

1.4. Is has been forked off after v1.4.4 (and before 1.5.0). The current version is 1.4.10.

> Wishing i had seen this a few months ago now.

I'm wishing it wasn't necessary 🙈