hjam40/Camera.MAUI #141: Android video is saved upside down

prashant-bhandari, 2024-02-21:
Hi, the video when recorded on android device is saved upside down.

Is there a work around for this issue?

janusw, 2024-02-22:
I believe this is fixed in release [1.4.6](https://www.nuget.org/packages/CameraMaui/1.4.6) (and upwards), built from [my fork](https://github.com/janusw/Camera.Maui).

prashant-bhandari, 2024-02-23:
It looks like there is only until 1.4.4 in nuget. Do I have to download another nuget? I have fixed it by adding the code directly and it is similar to yours with 270 degree rotation for front camera.

janusw, 2024-02-23:
> It looks like there is only until 1.4.4 in nuget. Do I have to download another nuget?

I posted the link to the nuget above: https://www.nuget.org/packages/CameraMaui/1.4.6

The name is "CameraMaui".

prashant-bhandari, 2024-02-24:
ah thanks for that. I did not pay attention on the name that minutely and thought it was the same

closed 2024-02-24 by prashant-bhandari