hjam40/Camera.MAUI #148: Any other alternative libraries?

romerotg, 2024-03-23:
I stumbled upon this library because using the built-in media picker isn't reliable, as sometimes the background app gets closed while the camera app is open. That is particularly true for Samsung devices. But this library isn't really working well for me. I get random crashes when switching the camera (also present on the demo) and can't get the view to the correct size. It always resizes it incorrectly when I try full screen, making the image preview stretched and blurry. Somehow it makes the preview wider than it should.

Is there another library that I can try as an alternative?

janusw, 2024-03-25:
I currently maintain my own fork, which fixes a few additional bugs. I don't know if it works any better for you, but you can try.

https://github.com/janusw/Camera.Maui
https://www.nuget.org/packages/CameraMaui

janusw, 2024-03-25:
> But this library isn't really working well for me. I get random crashes when switching the camera (also present on the demo)

You should open an issue for that and provide the backtrace, a way to reproduce (if possible), and the device/OS on which you see this ...

thomasgalliker, 2024-09-10:
My 5 cents:
https://www.nuget.org/packages/CameraScanner.Maui
https://github.com/thomasgalliker/CameraScanner.Maui

romerotg, 2024-09-10:
I ended implementing my own camera view