hjam40/Camera.MAUI #149: could not find file license.md

fabioikomotta, 2024-03-25:
I can't run the sample because this error.
Any sugestion?

CryptKat, 2024-03-26:
Easiest way - copy LICENSE.md from root folder to Camera.MAUI folder.

fabioikomotta, 2024-03-26:
The file is there but doesn't compile because app don't find it.
I tried set as Resource/Copy if newer but didn't works.

[1452173488405_logo_90x90.png]

Fabio Iko Motta
+5511 99487 6406
AKTTOM Sistemas
www.akttomsistemas.com.br<http://www.akttomsistemas.com.br>

renelaerke, 2024-06-16:
I've created a pull-request which fixes this problem.

The issue is that some references (inside the .csproj file) are still referencing the file "license.md" but the file has moved and is renamed to "../LICENSE.md"

With a little luck the pull-request is accepted and gets merged soon.