hjam40/Camera.MAUI #152: Barcode not shown up

KannanKrish, 2024-03-31:
Camara Maui version 1.5.1 QR Code not shown up.

Version 1,4,4 QR Code working fine.

bobwhitten, 2024-04-22:
yeah, I see the same thing.   I'm using this ONLY to display QR code.  Rolled back to 1.4.4 and see the QR code.

matthiasrijntjes, 2024-05-18:
> Camara Maui version 1.5.1 QR Code not shown up.

Same for me, and i also had to roll back to version 1.4.4 wat solved the problem for now.

melihercan, 2024-06-17:
Same here; I can detect barcodes with <= 1.4.4. 
Can this library detect QR codes? It cannot detect this one, even with 1.4.4.
[image]

matthiasrijntjes, 2024-06-17:
Yes, with 1.5.1 you can detect QR codes. But the configuration is a little bit different as in version 1.4.4. To use barcode detection with Camera.MAUI version 1.5.1 you also have to include the Camera.MAUI.ZXing Nuget package . Then in your sourcecode you have to specify the BarCodeDecoder property of your cameraview as shown in the screenshots and don't forget the Barcodeformat.QR code in PossibleFormats!

[image]
[image]

melihercan, 2024-06-17:
Hi @matthiasrijntjes, thanks for the response.  
I followed the GitHub page instructions as you pointed out above, and all of the configurations are in place. But still, no joy.

I just wanted to ask you this. Camera.MAUI.Test from the master branch should work, right? I gave it a try, and neither a bar code nor a QR code was detected. I have tried both on Android and Windows.

mrijntjes, 2024-06-18:
Hi @melihercan, i've not yet tested the original Camera.MAUI.Test which is currently based on .net 7.0. I've made an new sample repository for you, with an fresh sample based on .net 8.0 and tested on a Windows laptop with frontcamera. The repository is available at [https://github.com/mrijntjes/CameraMaui151sample](https://github.com/mrijntjes/CameraMaui151sample) with a sample QR code inside issue 1 on [https://github.com/mrijntjes/CameraMaui151sample/issues/1](https://github.com/mrijntjes/CameraMaui151sample/issues/1). If you still experience some problems then I suggest you to open or attribute to a new issue while this issue #152 was made for QRcode generating.

melihercan, 2024-06-18:
@mrijntjes Thanks for the prompt response. I will give it a try tomorrow and report the results here.
Thanks again.

melihercan, 2024-06-19:
@mrijntjes I have tested the provided repo both on Android and Windows. 
Android works :)
[image]
[image]

melihercan, 2024-06-19:
There is something wrong with the Windows settings. The following errors are observed:
If you can fix it, this will be a great reference app for QR detection. 
Thanks for your hard work.
[image]
[image]

Baroonn, 2024-09-11:
Had this same issue. It turned out to be my phone camera quality. My phone could scan 2D codes like QRCodes with the version 1.5.1 but it wouldn't detect 1D codes. 

Tried my laptop camera and it worked perfectly.