hjam40/Camera.MAUI #157: Slow Barcode Reading

garcia252, 2024-05-03:
I am experiencing very slow barcode reading for both version 1.4.4 and 1.5.1. First it was not working at all and then got it working by setting the WidthRequest to a fixed value as suggested by a different issue here on github.

Does anyone know how to get good speed for the barcode reading (QR and 1-D Codes)? Right now the user needs to target the code for several seconds before the app picks it up. And that is in Release mode build with a Samsung Galaxy S21. Also tried on a newer iPad and same issue.

strajk-, 2024-06-03:
> I am experiencing very slow barcode reading for both version 1.4.4 and 1.5.1. First it was not working at all and then got it working by setting the WidthRequest to a fixed value as suggested by a [...]

Current version is completely broken as far as detection is concerned, you're better off downgrading to 1.2.1 which doesn't require Camera.MAUI.ZXing, scans with that version are almost instant for me, some people completely abandoned this repo and moved on to others which I haven't tried yet since the downgrade was enough (for now).

skane526, 2024-06-06:
@strajk- I attempted to downgrade to 1.3.5 and was unable to find any references on how to setup the bar code scanning. When I go to the readme from the 1.3.5 repo it still requires ZXingHelper in the code.