hjam40/Camera.MAUI #161: Can't focus if subject is too near from the camera (for iPhone14 Pro and above models)

anjosantos, 2024-05-09:
I tested it in my iPhone15 Pro Max and it also happens with iPhone 14 Pro Maxes. Whenever you are trying to take pictures very close to the subject, it does not AutoFocus. I also tried using the ForceAutoFocus() method but still no luck. I think the ultra wide lens or the macro mode is still not supported in this library? Or at least for 14 Pro Maxes and 15 Pro Maxes?

Also the minimum Zoom Factor is always 1 for the Camera? So how can we try to set it to 0.5f ZoomFactor, to use the wide lens? Thank you!

Hugol13, 2024-05-09:
I had that problem and the only solution I found was to put two buttons in the view to control the plugin zoonFactor property to adjust the image quality.

anjosantos, 2024-05-10:
@Hugol13 you mean, whenever you want to capture in a macro (close to subject), you created buttons to zoom in further (then of course distancing your camera/phone farther away from the subject since you are zooming in)? Thanks!

renelaerke, 2024-06-15:
On an iPhone you have to activate  the correct camera or camera combination defined in the native apis to be able to capture with the correct camera.

An iPhone 14 Pro (my device) has the back facing optical lenses but by default this component solely activates the standard one. 

This results in you only being able to capture good quality images in the most normal use cases. Objects very close or far away is not handles well by this default.

By activating the correct camera or a "combination" of multiple cameras you will be able to take far better quality pictures. Unfortunately the author of this package however must make some code changes to make this happen.

Earlier today I issued a pull-request that mitigates this problem. Let's see how quickly @hjam40 accepts the proposed changes.

Btw for crisp closeups You should go for the "ultrawideangle" camera.

ngocnnt, 2024-09-13:
Hi @renelaerke , I am facing the same issue, can you guide me to fix it. I have added your pull-request "var deviceDescoverySession = AVCaptureDeviceDiscoverySession.Create(new AVCaptureDeviceType[] { AVCaptureDeviceType.BuiltInWideAngleCamera, AVCaptureDeviceType.BuiltInUltraWideCamera, AVCaptureDeviceType.BuiltInDualWideCamera, AVCaptureDeviceType.BuiltInTripleCamera, AVCaptureDeviceType.BuiltInTelephotoCamera }, AVMediaTypes.Video, AVCaptureDevicePosition.Unspecified);" to the code in the library but when I test it on iPhone 15 pro, it still doesn't focus if subject is too near from the camera.
Thank you.

Light407, 2024-11-21:
Any update on this PR @renelaerke @hjam40 ?