hjam40/Camera.MAUI #164: Resolution poor

Slordo, 2024-06-13:
I'm trying to get images to the highest resolution.
In this way:

var result = await cameraView.StartCameraAsync(cameraView.Camera.AvailableResolutions.OrderByDescending(size => size.Width * size.Height).FirstOrDefault());
if (result != CameraResult.Success)

      ResultLabel.Text = "Error starting camera";
      return;
}
cameraView.ForceAutoFocus();
Stream stream = cameraView.SnapShotStream;

My phone (Samsung S23) can take photo 4080x3060, but I take stream 1650x1050.
When I change the dimension of cameraview in my xaml file, the stream change resolution.
Actually I have this xaml:

<Grid>
<cv:CameraView x:Name="cameraView" HeightRequest="550" WidthRequest="350" HorizontalOptions="FillAndExpand" VerticalOptions="FillAndExpand" CamerasLoaded="cameraView_CamerasLoaded" />
</Grid>

Where do I make the mistake?
Thanks.

Slordo, 2024-06-17:
Today I realized something. The actual resolution is the (cv:CameraView) resolution multiplied by 3.
1650x1050 = (550x350) * 3
Anyone have an idea?
Thanks.

CRoederTND, 2024-10-22:
Maybe a bit late.... Try to change the camera, I also have a S23 and recieve two front and two back cameras from this plugin. One front and one back camera has a poor resolution, just swp to the one with the better resolution.